dlbeer/mspdebug #1: How to add new targets?

skalidindi3, 2015-07-09:
The target MCU is an MSP4305989, which does not appear to be in the `fet_olimex_db.c` file.

```
$ sudo ./mspdebug olimex -j -d /dev/cu.usbmodem000006A1 
MSPDebug version 0.23 - debugging tool for MSP430 MCUs
Copyright (C) 2009-2015 Daniel Beer <dlbeer@gmail.com>
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Chip info database from MSP430.dll v3.3.1.4 Copyright (C) 2013 TI, Inc.

Trying to open /dev/cu.usbmodem000006A1 at 115200 bps...
Resetting Olimex command processor...
Initializing FET...
FET protocol version is 20000007
Set Vcc: 3000 mV
Configured for JTAG (2)
Sending reset...
Using Olimex identification procedure
Device ID: 0xab81
fet: can't find device in DB
fet: identify failed
Trying again...
Resetting Olimex command processor...
Initializing FET...
FET protocol version is 20000007
Set Vcc: 3000 mV
Configured for JTAG (2)
Using Olimex identification procedure
Device ID: 0xab81
fet: can't find device in DB
fet: identify failed
```

dlbeer, 2015-07-09:
On Wed, Jul 08, 2015 at 05:43:52PM -0700, Sai Kalidindi wrote:
> The target MCU is an MSP4305989, which does not appear to be in the `fet_olimex_db.c` file.

Hi Sai,

Have you tried using the --fet-force-id flag to attempt to use a
configuration for a similar device? For example:

    mspdebug <flags> --fet-force-id MSP430FR5969

You can get a list of supported devices with the --fet-list option. Once
you find one that works, that can be adapted to add a configuration for
your chip.

Cheers,
Daniel

-- 
Daniel Beer <dlbeer@gmail.com> http://dlbeer.co.nz/
PGP: BA6E 0B26 1F89 246C E3F3  C910 1E58 C43A 160A 553B

skalidindi3, 2015-07-09:
Hi Daniel,

Thanks for the super speedy reply! I had not thought of doing that with the `5969`. I had tried `--fet-force-id MSP430FR5989` before and ran into errors:
```
$ sudo ./mspdebug olimex -j -d /dev/cu.usbmodem000006A1 --fet-force-id MSP430FR5989
MSPDebug version 0.23 - debugging tool for MSP430 MCUs
Copyright (C) 2009-2015 Daniel Beer <dlbeer@gmail.com>
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Chip info database from MSP430.dll v3.3.1.4 Copyright (C) 2013 TI, Inc.

Trying to open /dev/cu.usbmodem000006A1 at 115200 bps...
Resetting Olimex command processor...
Initializing FET...
FET protocol version is 20000007
Set Vcc: 3000 mV
Configured for JTAG (2)
Sending reset...
Using Olimex identification procedure
fet: no such device: MSP430FR5989
fet: identify failed
Trying again...
Resetting Olimex command processor...
Initializing FET...
FET protocol version is 20000007
Set Vcc: 3000 mV
Configured for JTAG (2)
Using Olimex identification procedure
fet: no such device: MSP430FR5989
fet: identify failed
```

Your suggestion seems to work:
```
$ sudo ./mspdebug olimex -j -d /dev/cu.usbmodem000006A1 --fet-force-id MSP430FR5969
MSPDebug version 0.23 - debugging tool for MSP430 MCUs
Copyright (C) 2009-2015 Daniel Beer <dlbeer@gmail.com>
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Chip info database from MSP430.dll v3.3.1.4 Copyright (C) 2013 TI, Inc.

Trying to open /dev/cu.usbmodem000006A1 at 115200 bps...
Resetting Olimex command processor...
Initializing FET...
FET protocol version is 20000007
Set Vcc: 3000 mV
Configured for JTAG (2)
Sending reset...
Using Olimex identification procedure
Device ID: 0xffff
  Code start address: 0x4400
  Code size         : 64512 byte = 63 kb
  RAM  start address: 0x1c00
  RAM  end   address: 0x23ff
  RAM  size         : 2048 byte = 2 kb
Device: MSP430FR5969
Number of breakpoints: 3
fet: FET returned error code 34 (Not supported by selected interface or interface is not initialized)
fet: warning: message C_IDENT3 failed
fet: FET returned NAK
warning: device does not support power profiling
Chip ID data: ab 81 21

Available commands:
    !               fill            power           setwatch_r      
    =               gdb             prog            setwatch_w      
    alias           help            read            simio           
    blow_jtag_fuse  hexout          regs            step            
    break           isearch         reset           sym             
    cgraph          load            run             verify          
    delbreak        load_raw        save_raw        verify_raw      
    dis             md              set             
    erase           mw              setbreak        
    exit            opt             setwatch        

Available options:
    color                       gdb_default_port            
    enable_bsl_access           gdb_loop                    
    enable_fuse_blow            gdbc_xfer_size              
    enable_locked_flash_access  iradix                      
    fet_block_size              quiet                       

Type "help <topic>" for more information.
Use the "opt" command ("help opt") to set options.
Press Ctrl+D to quit.

(mspdebug) 
```

Before you close out this issue, would you mind explaining what pitfalls there might be to forcing the ID to a different chip? i.e. different code size, RAM size, etc?

Thanks again for all your help!

Sai

dlbeer, 2015-07-09:
On Thu, Jul 09, 2015 at 11:07:23AM -0700, Sai Kalidindi wrote:
> Before you close out this issue, would you mind explaining what pitfalls there might be to forcing the ID to a different chip? i.e. different code size, RAM size, etc?
> 
> Thanks again for all your help!

Hi Sai,

I've never come across anything too serious, but it's possible that you
might run into programming problems if you specify a chip with more RAM
than your actual chip (chip RAM is used for buffering). In any case,
there is a "verify" command that you can use if you want to be sure.

Cheers,
Daniel

-- 
Daniel Beer <dlbeer@gmail.com> http://dlbeer.co.nz/
PGP: BA6E 0B26 1F89 246C E3F3  C910 1E58 C43A 160A 553B

skalidindi3, 2015-07-20:
Hey Daniel,

Sorry for the intermission here. Forcing the FET ID seems to be blocking me from flashing the MCU:

```
(mspdebug) prog ../../main.hex 
Erasing...
Programming...
Writing 4096 bytes at 4400...
fet: FET returned error code 7 (Could not write device memory)
fet: failed to write to 0x4400
ihex: error on line 259
(mspdebug) verify
prog: you need to specify a filename
(mspdebug) verify ../../main.hex 
Verifying 4096 bytes at 4400...
ERROR: mismatch at 4400 (read ff, expected 10)
ihex: error on line 259
```

dlbeer, 2015-07-22:
On Mon, Jul 20, 2015 at 01:48:32PM -0700, Sai Kalidindi wrote:
> Sorry for the intermission here. Forcing the FET ID seems to be blocking me from flashing the MCU:
>
> ```
> (mspdebug) prog ../../main.hex
> Erasing...
> Programming...
> Writing 4096 bytes at 4400...
> fet: [...]

Hi Sai,

Have you tried using a chip whose flash range is a superset of the one
you're using?

If you still have no luck with that method, let me know and I'll send
you some patches to try out.

Cheers,
Daniel

-- 
Daniel Beer <dlbeer@gmail.com> http://dlbeer.co.nz/
PGP: BA6E 0B26 1F89 246C E3F3  C910 1E58 C43A 160A 553B