dlbeer/mspdebug #3: unclear how to set simulator flash and RAM sizes

adicarlo, 2015-11-04:
I'm trying to make use of the simulator in the most correct way possible, but I'm finding no way to specify the size of the RAM area (e.g., 128 bytes or 256 bytes) nor the flash size (e.g., 2k to 8k).

Possible specifying these sizes are irrelevant to using the simulator, I'm certainly able to proceed with stock values.  It just bothers me there's no way that I can see....

dlbeer, 2015-11-04:
On Wed, Nov 04, 2015 at 02:21:52PM -0800, Adam Di Carlo wrote:
> I'm trying to make use of the simulator in the most correct way
> possible, but I'm finding no way to specify the size of the RAM area
> (e.g., 128 bytes or 256 bytes) nor the flash size (e.g., 2k to [...]

Hi Adam,

There is no way to set these values in the simulator. Aside from the
small section of IO memory space, everything else in the 16-bit address
space is treated as if it were RAM.

Cheers,
Daniel

-- 
Daniel Beer <dlbeer@gmail.com> http://dlbeer.co.nz/
PGP: BA6E 0B26 1F89 246C E3F3  C910 1E58 C43A 160A 553B

adicarlo, 2015-11-05:
Daniel Beer <notifications@github.com> writes:

> On Wed, Nov 04, 2015 at 02:21:52PM -0800, Adam Di Carlo wrote:
>> I'm trying to make use of the simulator in the most correct way
>> possible, but I'm finding no way to specify the size of the RAM area
>> (e.g., 128 bytes or 256 bytes) nor the flash size (e.g., 2k to 8k).
>
> There is no way to set these values in the simulator. Aside from the
> small section of IO memory space, everything else in the 16-bit
> address space is treated as if it were RAM.

Thanks for the reply.  Correct me if I'm wrong, but that implies that a
number of operations which might fire a reset on a real MSP430 wouldn't
not fire this reset in the simulator (such as writing to program
memory), correct?

If so, that's the bug (or missing feature) I'm trying to highlight.
If its interesting, I verify this against a real MPU vs the simulator
to confirm issue...

adicarlo, 2015-11-23:
Well, no response, so I guess that's not interesting.   I'll close the bug since its INVALID as I read it...

closed 2015-11-23 by adicarlo

dlbeer, 2015-11-23:
Hi Adam,

Sorry for not replying -- Gmail ate the last notification email for some reason.

But yes, what you say is correct. The simulator is far more tolerant of bad program behaviour than a real chip. I'd originally written the simulator mainly as a tool for reverse-engineering. I'd be happy to accept patches to make the behaviour match more closely.

Cheers,
Daniel