dlbeer/mspdebug #4: unable to step device after reset

adicarlo, 2015-11-23:
I'm unable to do more than one 'step' command in mspdebug after issuing a reset.

Bug reproduced on version 0.22 and 0.23 as of GIT 38ea6143cdf9e63a8f200af25b2d4ec39d62d066

Evidence from console:

> mspdebug rf2500
MSPDebug version 0.22 - debugging tool for MSP430 MCUs
Copyright (C) 2009-2013 Daniel Beer <dlbeer@gmail.com>
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Trying to open interface 1 on 072
rf2500: warning: can't detach kernel driver: No data available
Initializing FET...
FET protocol version is 30001000
Set Vcc: 3000 mV
Configured for Spy-Bi-Wire
fet: FET returned error code 4 (Could not find device or device not supported)
fet: command C_IDENT1 failed
Using Olimex identification procedure
Device ID: 0x2452
  Code start address: 0xe000
  Code size         : 8192 byte = 8 kb
  RAM  start address: 0x200
  RAM  end   address: 0x2ff
  RAM  size         : 256 byte = 0 kb
Device: MSP430G2xx2
Number of breakpoints: 2
fet: FET returned NAK
warning: device does not support power profiling
Chip ID data: 24 52
(mspdebug) reset
(mspdebug) step
    ( PC: 0f806)  ( R4: 04fff)  ( R8: 0fade)  (R12: 00079)  
    ( SP: 00280)  ( R5: 00000)  ( R9: 07f7f)  (R13: 09fff)  
    ( SR: 00000)  ( R6: 00000)  (R10: 0efff)  (R14: 0bded)  
    ( R3: 00000)  ( R7: 0ffa3)  (R11: 0ff4b)  (R15: 0fe03)  
0xf806:
    0f806: 31 40 80 02               MOV     #0x0280, SP
    0f80a: b0 12 94 f8               CALL    #0xf894
    0f80e: b0 12 a2 f8               CALL    #0xf8a2
    0f812: f2 42 29 00               MOV.B   #0x0008, &0x0029
(mspdebug) step
fet: FET returned error code 16 (Could not single step device)
fet: failed to restart CPU

dlbeer, 2015-11-23:
On Sun, Nov 22, 2015 at 08:51:25PM -0800, Adam Di Carlo wrote:
> I'm unable to do more than one 'step' command in mspdebug after issuing a reset.
>
> Bug reproduced on version 0.22 and 0.23 as of GIT 38ea6143cdf9e63a8f200af25b2d4ec39d62d066
>
> Evidence from console: [...]

Hi Adam,

Do you know whether this bug is present in older versions? Does
single-stepping work if you *don't* issue a reset?

Also, if you happen to have the library handy, would you mind trying
with the tilib driver?

Cheers,
Daniel

-- 
Daniel Beer <dlbeer@gmail.com> http://dlbeer.co.nz/
PGP: BA6E 0B26 1F89 246C E3F3  C910 1E58 C43A 160A 553B

adicarlo, 2015-11-23:
> Do you know whether this bug is present in older versions?

I'm pretty sure I remember it from 0.21 as well.  I tried to rollback to git 5477cd78d87bb061f54766efdceb988ddcdd6ea5 and compile, but I'm getting a compilation error on util/sport.c:
util/sport.c: In function â€˜sport_readâ€™:
util/sport.c:162:10: error: variable â€˜tvâ€™ has initializer but incomplete type
   struct timeval tv = {

amongst others....

I'll see about how to use the TI libraries...

> Does single-stepping work if you *don't* issue a reset?

It does indeed!

adicarlo, 2015-11-23:
Sadly I'm unable to attempt this with tilib:

 > mspdebug tilib
MSPDebug version 0.22 - debugging tool for MSP430 MCUs
Copyright (C) 2009-2013 Daniel Beer <dlbeer@gmail.com>
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

MSP430_GetNumberOfUsbIfs
No unused FET found.

I did manage to reproduce the problem in mspdebug 0.19, using 'mspdebug rf2500'.  It shows the problem just as version 0.22 and the latest do -- stepping works fine, but after reset, you can only step once.

dlbeer, 2015-11-23:
On Mon, Nov 23, 2015 at 07:16:05AM -0800, Adam Di Carlo wrote:
> I'm pretty sure I remember it from 0.21 as well.  I tried to rollback to git 5477cd78d87bb061f54766efdceb988ddcdd6ea5 and compile, but I'm getting a compilation error on util/sport.c:
> util/sport.c: [...]

Hmm... I don't have your chip here, but if you want to try something,
I'd suggest messing with the reset type. In fet_ctl(), in
drivers/fet_core.c, there is a switch statement. The case for
DEVICE_CTL_RESET is where the reset is issued.

There is a parameter, curently set to FET_RESET_ALL, which controls how
the reset is done. These are defined near the top of the file. What
happens if you change it to FET_RESET_RST (or something else)?

Cheers,
Daniel

-- 
Daniel Beer <dlbeer@gmail.com> http://dlbeer.co.nz/
PGP: BA6E 0B26 1F89 246C E3F3  C910 1E58 C43A 160A 553B

adicarlo, 2015-11-24:
Cool, I'll try that out!  Busy right now but I'll frob around at this and see what I can do...

adicarlo, 2015-11-24:
Tried FET_RESET_RST -- no go.

Ditto FET_RESET_PUC, FET_RESET_VCC -- no joy.

I want to make clear that reset is not totally broken.  For instance, I can do:

  reset
  run

and all works fine.  However, if I Ctrl-C and then try to step, I'll get error 16.  So, only the step which is run after another command fails.

Let me know what else we can try.  Also, I'm available on IRC, I'm 'adicarlo' on freenode.

Do you have a launchpad to work with?  If not, I could buy you one and send it to you direct!  They're cheap....

dlbeer, 2015-11-25:
On Tue, Nov 24, 2015 at 03:49:18PM -0800, Adam Di Carlo wrote:
> Tried FET_RESET_RST -- no go.
>
> Ditto FET_RESET_PUC, FET_RESET_VCC -- no joy.
>
> I want to make clear that reset is not totally broken.  For instance, I can do:
>
>   reset
>   run
>
> and all works fine. [...]

Hi Adam,

That's quite odd. Does it make any different if you try "regs" between
"reset" and "step"?

I do have a launchpad here, but my main constraint is time,
unfortunately. I'll do what I can to help, but I might not be able to
investigate too much directly.

Cheers,
Daniel

-- 
Daniel Beer <dlbeer@gmail.com> http://dlbeer.co.nz/
PGP: BA6E 0B26 1F89 246C E3F3  C910 1E58 C43A 160A 553B

adicarlo, 2015-11-25:
> That's quite odd. Does it make any different if you try "regs" between
> "reset" and "step"?

It does!  So my original recipe, still happening, is:
  reset
  step
  step -- bewm

But if I insert 'regs', the first step will blow up:
  reset
  regs
  step -- bewm

Another clue which might be interesting:
  reset
  step
  step -- bewm
  reset
  step -- works!!
  step -- bewm

Always the 2nd operation after the reset!   Although some operations work over and over:
  reset
  regs
  regs
Well, regs always works.

Also, this works:
  reset
  run

Also this works, although I expected it to fail:
  reset
  step
  run

Note that all this behavior is 100% consistent.

dlbeer, 2015-11-25:
On Wed, Nov 25, 2015 at 09:33:05AM -0800, Adam Di Carlo wrote:
> It does!  So my original recipe, still happening, is:
>   reset
>   step
>   step -- bewm
>
> But if I insert 'regs', the first step will blow up:
>   reset
>   regs
>   step -- bewm
>
> Another clue which might be [...]

Hmm. Ok, just a semi-wild guess, but what if you modify cmd_step() so
that it calls device_ctl(DEVICE_CTL_HALT) after doing the single-step
(DEVICE_CTL_STEP)?

Cheers,
Daniel

-- 
Daniel Beer <dlbeer@gmail.com> http://dlbeer.co.nz/
PGP: BA6E 0B26 1F89 246C E3F3  C910 1E58 C43A 160A 553B

adicarlo, 2015-11-26:
Well, I'm just cargo culting here, not really sure if this is what you asked for, but this is what I tried:

diff --git a/drivers/fet_core.c b/drivers/fet_core.c
index d89f607..9016da1 100644
--- a/drivers/fet_core.c
+++ b/drivers/fet_core.c
@@ -703,6 +703,12 @@ int fet_ctl(device_t dev_base, device_ctl_t action)
 		if (do_run(dev, FET_RUN_STEP) < 0)
 			return -1;
 
+		power_end(dev);
+		if (fet_proto_xfer(&dev->proto, C_STATE, NULL, 0, 1, 1) < 0) {
+			printc_err("fet: failed to halt CPU\n");
+			return -1;
+		}
+
 		for (;;) {
 			device_status_t status = fet_poll(dev_base);
 

But sadly, no change in behavior can be detected.

dlbeer, 2015-11-26:
On Wed, Nov 25, 2015 at 08:48:25PM -0800, Adam Di Carlo wrote:
> Well, I'm just cargo culting here, not really sure if this is what you asked for, but this is what I tried:
>
> diff --git a/drivers/fet_core.c b/drivers/fet_core.c
> index d89f607..9016da1 100644
> --- [...]

What if you put it after the poll loop (for (;;) { ... })?

Failing that, I'm a bit stumped. I have no documentation for the FET
protocol except what I've tried and got working.

-- 
Daniel Beer <dlbeer@gmail.com> http://dlbeer.co.nz/
PGP: BA6E 0B26 1F89 246C E3F3  C910 1E58 C43A 160A 553B

adicarlo, 2015-11-26:
> What if you put it after the poll loop (for (;;) { ... })?

Aye, tried that, no luck.

> Failing that, I'm a bit stumped. I have no documentation for the FET
> protocol except what I've tried and got working.

Well, I'm pretty new to this sort of debugging, but I'm willing to try to take a look.  Is possible to get me started with a citation of the relevant document/page ?