dlbeer/mspdebug #5: Compatibilty with MSP Debug Stack 3.5 on 64-bit systems

inguin, 2016-01-15:
With the 3.5 release of the MSP Debug Stack TI changed the signature of most exported functions: Older versions used the type long for most integer parameters, new versions use int32_t. As a result the new library is incompatible with mspdebug on 64-bit systems where longs are 64 bits wide.

I'm not sure what the best way to deal with that issue is. We could either change the function signatures in tilib.c to match the new API, effectively requiring all users to upgrade the the new DLL. Or we could try to detect the API version at runtime and conditionally use the old or new signatures.

The API version could be detected by calling the MSP430_GetNumberOfUsbIfs() function: We could initialize a long to ~0 and pass it to that function. On a 64-bit system the new library will initialize only the lower 32 bits, leaving the upper half at all ones.

dlbeer, 2016-01-15:
On Fri, Jan 15, 2016 at 07:37:04AM -0800, Ingo van Lil wrote:
> With the 35 release of the MSP Debug Stack TI changed the signature of most exported functions: Older versions used the type long for most integer parameters, new versions use int32_t As a result [...]

Hi Ingo,

Have you tried this with the latest git version? There is already a pair
of wrappers which deals with the differing 64-bit ABIs, and should work
roughly as you've described. If this doesn't work for you, please send
me a transcript of the output.

Cheers,
Daniel

-- 
Daniel Beer <dlbeer@gmail.com> http://dlbeer.co.nz/
PGP: BA6E 0B26 1F89 246C E3F3  C910 1E58 C43A 160A 553B

inguin, 2016-01-15:
Hmm, the code in the repository definitely looks different than what I debugged this afternoon. I can't check now, but most likely the checkout on my office PC still points to the old repository at SourceForge.

Sorry for the false alarm!

closed 2016-01-15 by inguin

inguin, 2016-01-18:
Confirmed, my clone was still pointing to the SourceForge repository. The latest version works like a charm. Sorry again!