dlbeer/mspdebug #9: Unable to use v0.23 with eZ430-RF2500 Dev tool board

tdelizy, 2016-01-27:
Hi,

Working with a ez430-RF2500 with mspdebug 0.23 I was unable to connect to the target. here is the log: 
```
$ mspdebug rf2500

MSPDebug version 0.23 - debugging tool for MSP430 MCUs
Copyright (C) 2009-2015 Daniel Beer <dlbeer@gmail.com>
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Chip info database from MSP430.dll v3.3.1.4 Copyright (C) 2013 TI, Inc.

Trying to open interface 1 on 014
Initializing FET...
FET protocol version is 30000002
Set Vcc: 3000 mV
Configured for Spy-Bi-Wire
Device ID: 0xf227
fet: unknown device
msg28_data: [0x12 bytes]
    f2 27 50 40 00 00 00 00 00 00 02 02 01 04 a1 61 
    80 00 
Using Olimex identification procedure
fet: missing info
fet: identify failed
Trying again...
Initializing FET...
FET protocol version is 30000002
Set Vcc: 3000 mV
Configured for Spy-Bi-Wire
Sending reset...
Device ID: 0xf227
fet: unknown device
msg28_data: [0x12 bytes]
    f2 27 50 40 00 00 00 00 00 00 02 02 01 04 a1 61 
    80 00 
Using Olimex identification procedure
fet: missing info
fet: identify failed
```

Tried on v0.22 and mspdebug can connect without facing the issue.

dlbeer, 2016-01-27:
On Wed, Jan 27, 2016 at 07:30:24AM -0800, Tristan Delizy wrote:
> Working with a ez430-RF2500 with mspdebug 023 I was unable to connect to the target here is the log:
> ```
> $ mspdebug rf2500
>
> MSPDebug version 023 - debugging tool for MSP430 MCUs
> Copyright (C) [...]

Hi Tristan,

What's the target chip on this board? Have you tried this with the
latest git version?

Cheers,
Daniel

-- 
Daniel Beer <dlbeer@gmail.com> http://dlbeer.co.nz/
PGP: BA6E 0B26 1F89 246C E3F3  C910 1E58 C43A 160A 553B

tdelizy, 2016-01-28:
Hi,
I was using the 0.23 tag, but I tried out of master today and got the same result. The processor on the board is the MSP430F2274 ([the ti board](http://www.ti.com/tool/ez430-rf2500)).

here is the log generated from the master built today : 
```
MSPDebug version 0.23 - debugging tool for MSP430 MCUs
Copyright (C) 2009-2015 Daniel Beer <dlbeer@gmail.com>
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Chip info database from MSP430.dll v3.3.1.4 Copyright (C) 2013 TI, Inc.

Trying to open interface 1 on 038
Initializing FET...
FET protocol version is 30000002
Set Vcc: 3000 mV
Configured for Spy-Bi-Wire
Device ID: 0xf227
fet: unknown device
msg28_data: [0x12 bytes]
    f2 27 50 40 00 00 00 00 00 00 02 02 01 04 a1 61 
    80 00 
Using Olimex identification procedure
fet: missing info
fet: identify failed
Trying again...
Initializing FET...
FET protocol version is 30000002
Set Vcc: 3000 mV
Configured for Spy-Bi-Wire
Sending reset...
Device ID: 0xf227
fet: unknown device
msg28_data: [0x12 bytes]
    f2 27 50 40 00 00 00 00 00 00 02 02 01 04 a1 61 
    80 00 
Using Olimex identification procedure
fet: missing info
fet: identify failed
```

regards,
Tristan

tomexplorador, 2016-05-28:
Hi tidelizy!
Were you able to fix it? I need help!
 Thanks!

dlbeer, 2016-05-28:
On Sat, May 28, 2016 at 03:26:09PM -0700, tomexplorador wrote:
> Hi tidelizy!
> Were you able to fix it? I need help!
>  Thanks!

Have either of you tried:

    mspdebug rf2500 --fet-force-id MSP430F2274

...assuming, of course, that the chip is a 2274.

Cheers,
Daniel

-- 
Daniel Beer <dlbeer@gmail.com> http://dlbeer.co.nz/
PGP: BA6E 0B26 1F89 246C E3F3  C910 1E58 C43A 160A 553B