dlbeer/mspdebug #10: strange register values

voxspox, 2016-02-18:
We experience strange register values (>16bit) using mspdebug with a TI Programmer and  tilib on a MSP430F1611 device. The libmsp430.so was compiled from MSPDebugStack_OS_Package (revisions.txt: Rev 3.5.1.001) and mspdebug is from the Ubuntu repository.

For testing we used a simple led blink application. After programming the LED blinks :) `run` and `step` is ok, but the register dump shows strange values. Additionally, setting a break point is not effective.

```
$ mspdebug tilib
MSPDebug version 0.23 - debugging tool for MSP430 MCUs
Copyright (C) 2009-2015 Daniel Beer <dlbeer@gmail.com>
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Chip info database from MSP430.dll v3.3.1.4 Copyright (C) 2013 TI, Inc.

MSP430_GetNumberOfUsbIfs
MSP430_GetNameOfUsbIf
Found FET: ttyACM0
MSP430_Initialize: ttyACM0
Firmware version is 30501001
MSP430_VCC: 3000 mV
MSP430_OpenDevice
MSP430_GetFoundDevice
Device: MSP430F1611 (id = 0x0071)
8 breakpoints available
MSP430_EEM_Init
Chip ID data: f1 6c

Available commands:
*snip*

Type "help <topic>" for more information.
Use the "opt" command ("help opt") to set options.
Press Ctrl+D to quit.

(mspdebug) prog bl.txt
Erasing...
Programming...
Writing  204 bytes at 4000...
Writing   32 bytes at ffe0...
Done, 236 bytes total
(mspdebug) run
Running. Press Ctrl+C to interrupt...
^C
    ( PC: 0408c)  ( R4: 0386a)  ( R8: 89f390)  (R12: 00002)
    ( SP: 00001)  ( R5: 00000)  ( R9: 00001)  (R13: 00000)
    ( SR: 038aa)  ( R6: 00000)  (R10: ebe1cea0)  (R14: 00000)
    ( R3: 00086)  ( R7: 01118)  (R11: 41330f)  (R15: 6b3b9740)
0x408c:
    0408c: 81 93 00 00               TST     0x0000(SP)
    04090: f5 27                     JZ      0x407c
    04092: fa 3f                     JMP     0x4088
    04094: 31 40 00 39               MOV     #0x3900, SP
    04098: b0 12 c4 40               CALL    #0x40c4
(mspdebug)
MSP430_Run
MSP430_Close
```

dlbeer, 2016-02-18:
On Thu, Feb 18, 2016 at 03:01:06AM -0800, knut wrote:
> We experience strange register values (>16bit) using mspdebug with a
> TI Programmer and  tilib on a MSP430F1611 device. The libmsp430.so was
> compiled from MSPDebugStack_OS_Package (revisions.txt: Rev [...]

Are you using a 64-bit system?

If so, you will need to build mspdebug from the git repository. The API
for the TI library changed on 64-bit systems in an incompatible way. The
latest mspdebug will detect the variant API and work correctly.

Cheers,
Daniel

-- 
Daniel Beer <dlbeer@gmail.com> http://dlbeer.co.nz/
PGP: BA6E 0B26 1F89 246C E3F3  C910 1E58 C43A 160A 553B

voxspox, 2016-02-19:
Yes, and yes this worked :)
thx!

closed 2016-02-19 by voxspox