dlbeer/mspdebug #19: Verify fails when invoked from linux shell

xhpohanka, 2016-11-09:
I'm loading firmware into MSP430F249 cpu. If I do programming from `mspdebug` interface calling `prog` and `verify` commands in sequence everything works well.

However when I do it with following command from shell
        
    mspdebug -j olimex "prog firmware.hex"

and verifying in next step

    mspdebug -j olimex "verify firmware.hex"

fails always on the same location

```
Chip ID data:
  ver_id:         49f2
  ver_sub_id:     0000
  revision:       04
  fab:            60
  self:           0000
  config:         00
Device: MSP430F249
Verifying 4096 bytes at 1200...
Verifying 4096 bytes at 2200...
Verifying 4096 bytes at 3200...
Verifying  816 bytes at 4200...
Verifying 4096 bytes at 5000...
Verifying 4096 bytes at 6000...
Verifying 4096 bytes at 7000...
Verifying 4096 bytes at 8000...
Verifying 4096 bytes at 9000...
Verifying 1110 bytes at a000...
Verifying   13 bytes at b604...
Verifying   24 bytes at b613...
Verifying  170 bytes at b632...
ERROR: mismatch at b692 (read bf, expected 00)
ihex: error on line 2185
```

dlbeer, 2016-11-09:
On Wed, Nov 09, 2016 at 03:06:30AM -0800, Jan Pohanka wrote:
> I'm loading firmware into MSP430F249 cpu. If I do programming from `mspdebug` interface calling `prog` and `verify` commands in sequence everything works well.
>
> However when I do it with following [...]

Hi Jan,

Have you tried a testing by manually programming and writing a word at
that specific location (using "erase", "md" and "mw")?

How old is the chip? Is it possible that the flash has gone bad?

Cheers,
Daniel

-- 
Daniel Beer <dlbeer@gmail.com> http://dlbeer.co.nz/
PGP: BA6E 0B26 1F89 246C E3F3  C910 1E58 C43A 160A 553B

xhpohanka, 2016-11-10:
Hi Daniel,
I'm doing described operation for loading firmware into small series of PCBs in production. Surprisingly this affects every piece of our hardware... 

thanks
Jan

dlbeer, 2016-11-10:
On Thu, Nov 10, 2016 at 01:29:01AM -0800, Jan Pohanka wrote:
> I'm doing described operation for loading firmware into small series
> of PCBs in production. Surprisingly this affects every piece of our
> hardware... 

Ok, but a small test with one board might help figure out what the
problem is. If you're able to do that I can look into it a bit further.

-- 
Daniel Beer <dlbeer@gmail.com> http://dlbeer.co.nz/
PGP: BA6E 0B26 1F89 246C E3F3  C910 1E58 C43A 160A 553B

xhpohanka, 2016-11-11:
Hi. Playing with md/mw has shown me the issue. I'm not author of the software loaded into MSP430 so I did not know that it writes some data on the flash. So after I had called prog command from Linux shell program in CPU started and overwrote mentioned location. Calling verify then simply cannot succeeded.
I have not found how to call multiple commands at one session so I added another one named `prog_verify` that does both (see https://github.com/xhpohanka/mspdebug/commits/prog_verify but it is just nasty hack)

thanks for this software

closed 2016-11-11 by xhpohanka

dlbeer, 2016-11-12:
On Fri, Nov 11, 2016 at 12:48:03AM -0800, Jan Pohanka wrote:
> Hi. Playing with md/mw has shown me the issue. I'm not author of the software loaded into MSP430 so I did not know that it writes some data on the flash. So after I had called prog command from [...]

Hi Jan,

No problem. You can non-interactively run multiple commands by
specifying them in separate command-line arguments. For example:

    mspdebug [driver options] "prog filename.hex" "verify filename.hex"

Cheers,
Daniel

-- 
Daniel Beer <dlbeer@gmail.com> http://dlbeer.co.nz/
PGP: BA6E 0B26 1F89 246C E3F3  C910 1E58 C43A 160A 553B