dlbeer/mspdebug #20: Implement jtgpio_rst and jtgpio_tst for devices with shared JTAG pins

johnp789, 2016-11-17:
I'd like to work with an MSP430F1232 using the gpio pins on a Raspberry Pi.  I think it requires 6 gpio pins on the Pi to drive the 4-wire JTAG pins along with the reset and test pins.  Would it be a difficult project to implement toggling the test and reset pins using the gpio driver, so that 4-wire JTAG on devices with shared JTAG pins can be supported?

dlbeer, 2016-11-17:
On Wed, Nov 16, 2016 at 08:19:36PM -0800, johnp789 wrote:
> I'd like to work with an MSP430F1232 using the gpio pins on a Raspberry Pi.  I think it requires 6 gpio pins on the Pi to drive the 4-wire JTAG pins along with the reset and test pins.  Would it be [...]

Hi John,

There is a "gpio" driver for this purpose (check the manual for
configuration options).

However, note that during flash memory programming, the MSP430 requires
a clock to be supplied within a certain frequency range (hundreds of
kHz). This is done in the driver by userspace bit-bashing. This seems to
work in a lot of cases, but it's not something I'd guarantee.

Cheers,
Daniel

-- 
Daniel Beer <dlbeer@gmail.com> http://dlbeer.co.nz/
PGP: BA6E 0B26 1F89 246C E3F3  C910 1E58 C43A 160A 553B

johnp789, 2016-11-17:
Right, I intended to use the gpio driver, but where do I associate gpio pins with the target device's test and reset pins?  I see only the four JTAG pins as -d options.  In the code, calls to set the test and reset pins are stubbed out with printf.

dlbeer, 2016-11-17:
On Thu, Nov 17, 2016 at 12:30:58PM -0800, johnp789 wrote:
> Right, I intended to use the gpio driver, but where do I associate gpio pins with the target device's test and reset pins?  I see only the four JTAG pins as -d options.  In the code, calls to set the test and reset pins are stubbed out with printf.  

Ah, I see. That doesn't appear to be implemented (I've never used the
gpio driver myself).

If you wanted to add the code to control those pins, I'd be happy to
accept a patch or pull request.

Cheers,
Daniel

-- 
Daniel Beer <dlbeer@gmail.com> http://dlbeer.co.nz/
PGP: BA6E 0B26 1F89 246C E3F3  C910 1E58 C43A 160A 553B

johnp789, 2016-11-19:
#21 adds support for these pins.  I've tested it with a Raspberry Pi Zero and an MSP430F449.

dlbeer, 2016-11-20:
Great, thanks for that!

closed 2016-11-20 by dlbeer