dlbeer/mspdebug #24: FET430UIF downgrade

xhpohanka, 2016-11-29:
Hi,
I have followed your instructions for downgrade (http://dlbeer.co.nz/mspdebug/tilib.html) and ended up with probably blank device that shows in Linux still with id 2047:0010 and `/dev/ttyACM0` node.

`tilib` driver wont initialize anymore

```
#mspdebug tilib --allow-fw-update
MSPDebug version 0.24 - debugging tool for MSP430 MCUs
Copyright (C) 2009-2016 Daniel Beer <dlbeer@gmail.com>
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Chip info database from MSP430.dll v3.3.1.4 Copyright (C) 2013 TI, Inc.

Using new (SLAC460L+) API
MSP430_GetNumberOfUsbIfs
MSP430_GetNameOfUsbIf
Found FET: ttyACM0
MSP430_Initialize: ttyACM0
tilib: MSP430_Initialize: Interface Communication error (error = 35)
tilib: device initialization failed
```

Do you please have any recommendation how to 'debrick' the programmer? I have tried msp430flasher for windows but it also is not able to initialize the device.

dlbeer, 2016-11-29:
On Tue, Nov 29, 2016 at 02:40:43AM -0800, Jan Pohanka wrote:
> Hi,
> I have followed your instructions for downgrade (http://dlbeer.co.nz/mspdebug/tilib.html) and ended up with probably blank device that shows in Linux still with id 2047:0010 and `/dev/ttyACM0` [...]

Hi Jan,

Those instructions were given to me by TI, although that was quite a few
firmware versions back.

If you open the device up, you may be able to get access to a BSL
interface to write new firmware.  Of course, you will need to grab an
older firmware image from somewhere and an older device to program with.
Email me (dlbeer@gmail.com) if you have any trouble with this.

Cheers,
Daniel

-- 
Daniel Beer <dlbeer@gmail.com> http://dlbeer.co.nz/
PGP: BA6E 0B26 1F89 246C E3F3  C910 1E58 C43A 160A 553B

xhpohanka, 2016-12-02:
I fixed it by reading V3 firmware from another FET430UIF and, thanks.

closed 2016-12-02 by xhpohanka