dlbeer/mspdebug #27: Get error "tilib_api: can't find libmsp430.so: libmsp430.so: wrong ELF class: ELFCLASS32"

isaacgr, 2017-01-04:
Get this error when I run `mspdebug tilib`. Is there anything I can do? libmsp430.so is a 32 bit file and I've tried putting it in /usr/lib, /usr/lib32 and some other paths but still the same issue. I would appreciate any assistance.

isaacgr, 2017-01-04:
I actually solved my issue with the instructions here: https://github.com/chintal/toolchains/blob/master/msp430-gcc-ti.md under the heading 'Installing 64-bit libmsp430.so v3.8'. Sadly my MSP still appears to be bricked (was trying to do a firmware upgrade) but I appreciate the tool!

isaacgr, 2017-01-04:
Really sorry for all the posts. Looks like I spoke too soon, just had to try a couple of times to get it. I now get about this far:
`MSPDebug version 0.24 - debugging tool for MSP430 MCUs
Copyright (C) 2009-2016 Daniel Beer <dlbeer@gmail.com>
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Chip info database from MSP430.dll v3.3.1.4 Copyright (C) 2013 TI, Inc.

Using new (SLAC460L+) API
MSP430_GetNumberOfUsbIfs
MSP430_GetNameOfUsbIf
Found FET: ttyACM0
MSP430_Initialize: ttyACM0
Firmware version is 30901602
MSP430_VCC: 3000 mV
MSP430_OpenDevice
tilib: MSP430_OpenDevice: Device database not loaded. (error = 98)
tilib: device initialization failed
`
Any ideas at this point? I'm using Code Composer and now the program actually compiles and loads onto the chip but still complains that I need to upgrade my firmware (which I am reluctant to do since that is what started all this in the first place).

dlbeer, 2017-01-04:
On Wed, Jan 04, 2017 at 05:18:13AM -0800, isaacgr wrote:
> Really sorry for all the posts. Looks like I spoke too soon, just had to try a couple of times to get it. I now get about this far:
> `MSPDebug version 0.24 - debugging tool for MSP430 MCUs
> Copyright [...]

That looks like the interface between mspdebug and the library itself is
working, but there is some issue between the library and the firmware.

Unfortunately you might have to try the firmware update -- either that,
or check with TI. The error codes quoted in the mspdebug output come
straight from TI's library.

Cheers,
Daniel

-- 
Daniel Beer <dlbeer@gmail.com> http://dlbeer.co.nz/
PGP: BA6E 0B26 1F89 246C E3F3  C910 1E58 C43A 160A 553B

isaacgr, 2017-01-05:
I took the risk and upgraded through Code Composer. It worked no problem. Thank you very much.

xhpohanka, 2017-02-15:
> tilib: MSP430_OpenDevice: Device database not loaded. (error = 98)

It seems that slac460s release of MSP430 library sources lacks device database initialization. It can be solved like in my fork https://github.com/xhpohanka/mspdebug/commit/a3254fb2b8285699387c8c82a28a51c060e3a268

dlbeer, 2017-02-16:
Cool -- did you want to put together a pull request for this change, or would you prefer if I picked out and applied the diff on its own?

xhpohanka, 2017-02-17:
I will prepare pull request.
There is strange thing that binary version of slac460s provided by TI does not need this, they probably released slightly different source...

dlbeer, 2017-02-20:
Yes, they might well have done. I think some of the very early releases might have been a little out-of-sync with source too.