dlbeer/mspdebug #33: cannot connect to TI ez430-chronos

esno, 2017-03-25:
Hi I tried to flash my TI chronos clock but everytime it ends up in `Connection Timed Out`
I tried current master, v0.22 and v0.24.

    $ mspdebug rf2500 
    MSPDebug version 0.22 - debugging tool for MSP430 MCUs
    Copyright (C) 2009-2013 Daniel Beer <dlbeer@gmail.com>
    This is free software; see the source for copying conditions.  There is NO
    warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

    Trying to open interface 1 on 007
    rf2500: warning: can't detach kernel driver: No such file or directory
    Initializing FET...
    FET protocol version is 30001000
    Set Vcc: 3000 mV
    Configured for Spy-Bi-Wire
    fet: FET returned error code 4 (Could not find device or device not supported)
    fet: command C_IDENT1 failed
    Using Olimex identification procedure
    fet: FET returned error code 4 (Could not find device or device not supported)
    fet: missing info
    fet: identify failed
    Trying again...
    Initializing FET...
    FET protocol version is 30001000
    Set Vcc: 3000 mV
    Configured for Spy-Bi-Wire
    Sending reset...
    rf2500: can't receive data: Connection timed out
    warning: fet: reset failed
    fet: FET returned error code 4 (Could not find device or device not supported)
    fet: command C_IDENT1 failed
    Using Olimex identification procedure
    fet: FET returned error code 4 (Could not find device or device not supported)
    fet: missing info
    fet: identify failed

It looks like the clock will be reseted. Display got off and on again when mspdebug comes to `sending reset...` but then it ends in connection timed out.

Any idea how to follow up this issue?

dlbeer, 2017-03-26:
On Sat, Mar 25, 2017 at 07:20:13AM -0700, esno wrote:
> Hi I tried to flash my TI chronos clock but everytime it ends up in `Connection Timed Out`
> I tried current master, v0.22 and v0.24.
>
>     $ mspdebug rf2500
>     MSPDebug version 0.22 - debugging tool [...]

I don't know much about the Chronos, but I would first double-check the
board power arrangements (see whether it expects power via the
debugger), and also check to see whether it is Spy-Bi-Wire and not JTAG.

Failing that, I would try using TI's libmsp430.so instead (use the
"tilib" driver with the latest git version of mspdebug). This should
support every device that TI have produced.

Cheers,
Daniel

-- 
Daniel Beer <dlbeer@gmail.com> http://dlbeer.co.nz/
PGP: BA6E 0B26 1F89 246C E3F3  C910 1E58 C43A 160A 553B

esno, 2017-03-26:
hi, thanks for information.
It looks like ez430-rs2500 supports only Spy-Bi-Wire and no JTAG.
Currently trying to compile libmsp to get a 64bit version. It's a hard job.

They messed around with dependencies and something is wrong with libboost_filesystem dependency

    /usr/bin/ld: /usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/libboost_filesystem.a(operations.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a shared object; recompile with -fPIC
    /usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/libboost_filesystem.a: error adding symbols: Bad value
    collect2: error: ld returned 1 exit status
    Makefile:133: recipe for target 'all' failed
    make: *** [all] Error 1

Let's see if I can finish the build to test tilib driver.
In the meantime I checked the flash procedure via a windows computer using the CCS and it worked.
But I would prefer a way using opensource and a linux computer.
I'll update this issue as soon as I finished the libmsp build.