dlbeer/mspdebug #34: mspdebug on macOS

eblot, 2017-04-25:
Hi,

RF2500s are HID USB devices.
Unfortunately macOS does not let libusb to access these devices. A so-called 'codeless' kernel extension would be required. However it should be signed, and Apple does not accept KEXT signing for open source projects (it seems it would have to be signed off by the manufacturer/owner, i.e. TI... Good luck with that)
https://github.com/libusb/libusb/wiki/FAQ#Does_libusb_support_USB_HID_devices

However, there seems to be an alternative: use the HID API to drive the HID device on macOS. I guess the best option would be to use the native HID framework to do so.

Meanwhile, here is a proof of concept that relies on HIDAPI that it nice enough to provide an API that is very close to libusb: https://github.com/eblot/mspdebug/tree/rf2500_hidapi

At least, we can use the excellent mspdebug tool on OSX / macOS with a Launchpad :-)

dlbeer, 2017-04-25:
That looks good! If you're reasonably confident that it works and hasn't broken anything, I'd be happy to merge this -- just create a pull request.

eblot, 2017-04-26:
This commit was only intended as a proof of concept :-)

I thought a simpler implementation could be based on direct calls to the IOHID native API rather than relaying on another dependency, just as BSLOSX does. 

I gave a try with native IOHID class (duplicating BSLOSX implementation), it also works.
Nevertheless the HIDAPI code is better (it uses a dedicated thread for the receive loop, resource and error management are more complete, etc.). In other words, I think HIDAPI is a cleaner solution, although it adds another dependency.

If this solution is ok for you, I'll try to find some spare time to ensure the changes do not impact Linux/FreeBSD ports and that GDB feature is also ok - for now I only tested flash programming. Let me know.

eblot, 2017-04-26:
It seems Ok. I fixed the serial number selection.

It builds and basically execute on Linux (Ubuntu 16.04 LTS x64) and FreeBSD (11.0 x64), although for the latter the path to support the _readline_ library seems to be missing (`-I/usr/local/include` & `-L/usr/local/lib`) but it does not seem to be related to my changes.

Single stepping on macOS using remote GDB seems to work (at least to the same extent as with Linux). I have not checked the Windows build ... I do not use it)

I'll send you a pull request. Thanks.

BTW I guess the FAQ should be updated for macOS (http://dlbeer.co.nz/mspdebug/faq.html#rf2500_osx) as this method does not work on recent releases of macOS (because of KEXT signing, AFAICT).

dlbeer, 2017-04-26:
Just merged now, and updated FAQ. Thanks again for implementing and testing this -- much appreciated.

closed 2017-04-26 by dlbeer

eblot, 2017-04-27:
You're welcome. HTH. Thanks for merging it.