dlbeer/mspdebug #36: tilib: broken for 0.24 and MSP-EXP430FR5969

daniel-k, 2017-05-11:
Hi,

I'm trying to get my [EXP-MSP430FR5969](http://www.ti.com/tool/msp-exp430fr5969) running and it looks like 0.24 is not working with `tilib` and the latest `mspds`. I install [mspdebug](https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=mspdebug) and [mspds](https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=mspds) on Archlinux via AUR, but only a downgrade of mspdebug to 0.23 yields a working debugger.

mspdebug 0.24 wants to update the firmware and then fails:

```bash
$ sudo mspdebug tilib --allow-fw-update
MSPDebug version 0.24 - debugging tool for MSP430 MCUs
Copyright (C) 2009-2016 Daniel Beer <dlbeer@gmail.com>
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Chip info database from MSP430.dll v3.3.1.4 Copyright (C) 2013 TI, Inc.

Using new (SLAC460L+) API
MSP430_GetNumberOfUsbIfs
MSP430_GetNameOfUsbIf
Found FET: ttyACM0
MSP430_Initialize: ttyACM0
FET firmware update is required.
Starting firmware update (this may take some time)...
Initializing bootloader...
Programming new firmware...
     4 percent done
    20 percent done
    36 percent done
    52 percent done
    52 percent done
    52 percent done
    52 percent done
    52 percent done
    68 percent done
    84 percent done
   100 percent done
Update complete
Done, finishing...
Initializing bootloader...
Programming new firmware...
     4 percent done
    20 percent done
    36 percent done
    52 percent done
    52 percent done
    52 percent done
    52 percent done
    52 percent done
    68 percent done
    84 percent done
   100 percent done
Update complete
Done, finishing...
Initializing bootloader...
Programming new firmware...
     4 percent done
    20 percent done
    36 percent done
    52 percent done
    52 percent done
    52 percent done
    52 percent done
    52 percent done
    68 percent done
    84 percent done
   100 percent done
Update complete
Done, finishing...
tilib: MSP430_FET_FwUpdate: MSP-FET / eZ-FET legacy module update failed (error = 75)
tilib: device initialization failed
```

Working downgrade to 0.23
```bash
$ sudo mspdebug tilib --allow-fw-update
MSPDebug version 0.23 - debugging tool for MSP430 MCUs
Copyright (C) 2009-2015 Daniel Beer <dlbeer@gmail.com>
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Chip info database from MSP430.dll v3.3.1.4 Copyright (C) 2013 TI, Inc.

MSP430_GetNumberOfUsbIfs
MSP430_GetNameOfUsbIf
Found FET: ttyACM0
MSP430_Initialize: ttyACM0
Firmware version is 4294967295
MSP430_VCC: 3000 mV
MSP430_OpenDevice
MSP430_GetFoundDevice
Device:  (id = 0x0000)
0 breakpoints available
MSP430_EEM_Init
Chip ID data: 03 00

Available commands:
    !               fill            power           setwatch_r      
    =               gdb             prog            setwatch_w      
    alias           help            read            simio           
    blow_jtag_fuse  hexout          regs            step            
    break           isearch         reset           sym             
    cgraph          load            run             verify          
    delbreak        load_raw        save_raw        verify_raw      
    dis             md              set             
    erase           mw              setbreak        
    exit            opt             setwatch        

Available options:
    color                       gdb_default_port            
    enable_bsl_access           gdb_loop                    
    enable_fuse_blow            gdbc_xfer_size              
    enable_locked_flash_access  iradix                      
    fet_block_size              quiet                       

Type "help <topic>" for more information.
Use the "opt" command ("help opt") to set options.
Press Ctrl+D to quit.

(mspdebug)
```

dlbeer, 2017-05-11:
The only version of mspdebug likely to work correctly with SLAC460S and up is the current master branch, so I wouldn't bother doing anything further with 0.23 and 0.24.

Unless there's some new incompatibility I'm unaware of, this looks like a problem between your board and the library. You might need to contact TI about this.

daniel-k, 2017-05-12:
Well at least the error messages change. In the following, I'm using current master and SLAC460S.

```bash
$ sudo mspdebug tilib --allow-fw-update
...

Using new (SLAC460L+) API
MSP430_GetNumberOfUsbIfs
MSP430_GetNameOfUsbIf
Found FET: ttyACM0
MSP430_Initialize: ttyACM0
FET firmware update is required.
Starting firmware update (this may take some time)...
Initializing bootloader...
Programming new firmware...
     4 percent done
     4 percent done
     4 percent done
     4 percent done
     4 percent done
     4 percent done
     4 percent done
    20 percent done
    20 percent done
    20 percent done
    20 percent done
    20 percent done
    36 percent done
    52 percent done
    52 percent done
    52 percent done
    52 percent done
    52 percent done
    68 percent done
    84 percent done
    84 percent done
    84 percent done
    84 percent done
    84 percent done
   100 percent done
Update complete
Done, finishing...
MSP430_VCC: 3000 mV
MSP430_OpenDevice
tilib: MSP430_OpenDevice: Device database not loaded. (error = 98)
```
So programming more or less works now I guess. But when I subsequently try to connect, I get:

```bash
$ sudo mspdebug tilib
...

Using new (SLAC460L+) API
MSP430_GetNumberOfUsbIfs
MSP430_GetNameOfUsbIf
Found FET: ttyACM0
MSP430_Initialize: ttyACM0
tilib: MSP430_Initialize: Interface Communication error (error = 35)
tilib: device initialization failed
```

And after a while also:

```bash
$ sudo mspdebug tilib
...

Using new (SLAC460L+) API
MSP430_GetNumberOfUsbIfs
No unused FET found.
```

Maybe this helps. There are people on the internet [1](http://blog.3mdeb.com/2015/12/08/using-platformio-with-ti-msp430-lunchpads/) [2](https://www.pabigot.com/msp430/exp430fr5969-launchpad-first-experiences/) who are using the exact same board with mspdebug, so I guess it's not that.

dlbeer, 2017-05-15:
Do older versions of the TI library work (pre-S)? If not, do you have access to a Windows machine where you can test Code Composer Studio to see if that works with your board?

daniel-k, 2017-05-15:
I finally got hold of [slac460r.zip](distcache.FreeBSD.org/ports-distfiles/msp430-debug-stack-3.8.0.1/slac460r.zip) which is non-trivial thanks to TI release policy, and it works now. So I can say that `S` definitely breaks mspdebug for TI Launchpad FR5969 for me.

spoorcc, 2017-05-18:
Thanks @daniel-k, I had the exact same problems, I used slac460r and it again worked for me (FR6989)

CrazyCasta, 2017-06-14:
@dlbeer : can we get a 0.25 out to handle this s issue? It would be nice if all the packages could get uploaded so I can stop running self compiled code from my home directory :)

dlbeer, 2017-06-15:
Yes, that'd be a good idea. I'm a bit tied up at the moment, but will add this to my to-do list. If I don't hear of anything significantly broken in the master branch within a week, I'll put together a release.