dlbeer/mspdebug #38: rf2500 is broken on macOS in v0.25

robertinant, 2017-08-03:
using the rf2500 driver on macOS in v0.25 results in the below error. It works OK in v0.24.
./mspdebug rf2500 --force-reset "prog Blink.ino.hex"
MSPDebug version 0.25 - debugging tool for MSP430 MCUs
Copyright (C) 2009-2017 Daniel Beer <dlbeer@gmail.com>
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Chip info database from MSP430.dll v3.3.1.4 Copyright (C) 2013 TI, Inc.

rf2500: failed to open RF2500 device

dlbeer, 2017-08-04:
On Thu, Aug 03, 2017 at 10:53:47AM -0700, robertinant wrote:
> using the rf2500 driver on macOS in v0.25 results in the below error. It works OK in v0.24.
> ./mspdebug rf2500 --force-reset "prog Blink.ino.hex"
> MSPDebug version 0.25 - debugging tool for MSP430 [...]

What happens if you try this as root? Can you try the same command with
sudo in front?

Cheers,
Daniel

-- 
Daniel Beer <dlbeer@gmail.com> http://dlbeer.co.nz/
PGP: BA6E 0B26 1F89 246C E3F3  C910 1E58 C43A 160A 553B

robertinant, 2017-08-07:
Same issue executing it as root.

dlbeer, 2017-08-07:
On Mon, Aug 07, 2017 at 08:10:39AM -0700, robertinant wrote:
> Same issue executing it as root.

Can you tell me the USB vendor and product IDs of your device? You can
apparently install "lsusb" via brew to do this, but perhaps there are
other ways in OS/X.

-- 
Daniel Beer <dlbeer@gmail.com> http://dlbeer.co.nz/
PGP: BA6E 0B26 1F89 246C E3F3  C910 1E58 C43A 160A 553B

themacks, 2017-10-18:
I'm having the same problem. Here is the USB info. Full data from USB Prober is attached. 
 
$ mspdebug --usb-list
Devices on bus 020:
    020:020 0451:f432 eZ430-RF2500 [serial: 96FF4204714E3A3A]

[USB Bus Probe.txt](https://github.com/dlbeer/mspdebug/files/1396429/USB.Bus.Probe.txt)

dlbeer, 2017-10-18:
On Wed, Oct 18, 2017 at 09:39:50PM +0000, themacks wrote:
> I'm having the same problem. Here is the USB info. Full data from USB Prober is attached. 
>  
> $ mspdebug --usb-list
> Devices on bus 020:
>     020:020 0451:f432 eZ430-RF2500 [serial: 96FF4204714E3A3A]
> 
> [USB Bus Probe.txt](https://github.com/dlbeer/mspdebug/files/1396429/USB.Bus.Probe.txt)

If you edit the Makefile, and comment out these two lines within the
OS/X-specific configuration:

      BSLHID_OBJ = transport/bslosx.o
      RF25000_OBJ = transport/rf2500hidapi.o

...and then do a "make clean" followed by "make", does that get things
working?

-- 
Daniel Beer <dlbeer@gmail.com> http://dlbeer.co.nz/
PGP: BA6E 0B26 1F89 246C E3F3  C910 1E58 C43A 160A 553B

themacks, 2017-10-19:
Removing both yields:
```
Undefined symbols for architecture x86_64:
  "_bslosx_open", referenced from:
      _loadbsl_open in loadbsl.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [mspdebug] Error 1
```

Removing only rf2500hidapi.o connects successfully. 
```
$ ./mspdebug rf2500
MSPDebug version 0.25 - debugging tool for MSP430 MCUs
Copyright (C) 2009-2017 Daniel Beer <dlbeer@gmail.com>
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Chip info database from MSP430.dll v3.3.1.4 Copyright (C) 2013 TI, Inc.

Trying to open interface 1 on 026
Initializing FET...
FET protocol version is 30066536
Set Vcc: 3000 mV
Configured for Spy-Bi-Wire
Device ID: 0xf201
  Code start address: 0xf800
  Code size         : 2048 byte = 2 kb
  RAM  start address: 0x200
  RAM  end   address: 0x27f
  RAM  size         : 128 byte = 0 kb
Device: F20x2_G2x2x_G2x3x
Number of breakpoints: 2
fet: FET returned NAK
warning: device does not support power profiling
Chip ID data:
  ver_id:         01f2
  ver_sub_id:     0000
  revision:       30
  fab:            40
  self:           0000
  config:         02
  fuses:          00
Device: F20x2_G2x2x_G2x3x
```

tgtakaoka, 2018-06-29:
Can you try the latest mspdebug on your environment again?
Your issue may be resolved by pull request #58.

trhodeos, 2018-09-09:
I was seeing this issue in .25, building from master with #58 included fixed it for me.

Euler1707, 2018-10-29:
Why did you guys break this for Mac OS, is a fix in the works?