dlbeer/mspdebug #39: Support for msp430fr2633?

Davito3, 2017-08-11:
Mspdebug does not seem to work with the msp430fr2633 using the captivate-pgrm board and ezfet. It does not appear to be in the mspdebug --fet-list

dlbeer, 2017-08-11:
On Thu, Aug 10, 2017 at 06:29:15PM -0700, Davito3 wrote:
> Mspdebug does not seem to work with the msp430fr2633 using the captivate-pgrm board and ezfet. It does not appear to be in the mspdebug --fet-list 

Which driver are you using? The list is for the "uif" and "rf2500"
drivers rather than the "ezfet" driver, which is somewhat experimental.

You will probably have more success using the "tilib" driver along with
libmsp430.so (the TI open-source library).

Cheers,
Daniel

-- 
Daniel Beer <dlbeer@gmail.com> http://dlbeer.co.nz/
PGP: BA6E 0B26 1F89 246C E3F3  C910 1E58 C43A 160A 553B

Davito3, 2017-08-11:
I'll give it a shot in the morning and let you know-mcu is at work currently thanks

Davito3, 2017-08-11:
where can i find the libmsp430.so it says it can't find it

Davito3, 2017-08-11:
got it working!

closed 2017-08-11 by Davito3