dlbeer/mspdebug #40: Fet returned error code 4 Olimex msp430 JTAG

simion96, 2017-10-12:
Hey Dan,

I've been trying to run mspdebug with the Olimex msp430 JTAG ISO 2, and haven't had much luck getting in.

Below is my dmesg output, everything seems fine to me..
`[  572.843938] usb 2-2.2: new full-speed USB device number 7 using uhci_hcd
[  573.144301] usb 2-2.2: New USB device found, idVendor=15ba, idProduct=0100
[  573.144304] usb 2-2.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  573.144306] usb 2-2.2: Product: MSP430 JTAGISO ver.2 
[  573.144308] usb 2-2.2: Manufacturer: OLIMEX LTD.
[  573.144309] usb 2-2.2: SerialNumber: 00002B5F
[  573.154028] cdc_acm 2-2.2:1.0: ttyACM0: USB ACM device
`

Full trace for running the connect command shown below:
`
sim@ubuntu:~$ sudo mspdebug -j -d /dev/ttyACM0 olimex-iso
MSPDebug version 0.25 - debugging tool for MSP430 MCUs
Copyright (C) 2009-2017 Daniel Beer <dlbeer@gmail.com>
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Chip info database from MSP430.dll v3.3.1.4 Copyright (C) 2013 TI, Inc.

Trying to open /dev/ttyACM0 at 200000 bps...
Resetting Olimex command processor...
Initializing FET...
FET protocol version is 20000007
Set Vcc: 3000 mV
Configured for JTAG (2)
Using Olimex identification procedure
fet: FET returned error code 4 (Could not find device or device not supported)
fet: missing info
fet: identify failed
Trying again...
Resetting Olimex command processor...
Initializing FET...
FET protocol version is 20000007
Set Vcc: 3000 mV
Configured for JTAG (2)
Using Olimex identification procedure
fet: FET returned error code 4 (Could not find device or device not supported)
fet: missing info
fet: identify failed
`

Any ideas? :(

dlbeer, 2017-10-12:
On Thu, Oct 12, 2017 at 07:46:07PM +0000, Simion Putina wrote:
> Hey Dan,
>
> I've been trying to run mspdebug with the Olimex msp430 JTAG ISO 2, and haven't had much luck getting in.
>
> Below is my dmesg output, everything seems fine to me..
> `[  572.843938] usb [...]

Hi Simion,

That looks like communication to the JTAG tool is working fine, but the
JTAG connection between the tool and the board is not.

Some things to check:

  * Are the JTAG connections all ok, and have you connected the right
    pins?

  * Is the chip powered? Are you supposed to be supplying power via
    JTAG, or does the board have its own supply (you're currently
    supplying 3V via JTAG, it looks like).

  * Is the interface actually JTAG, or Spi-Bi-Wire?

  * Do you have a board fault?

Cheers,
Daniel

-- 
Daniel Beer <dlbeer@gmail.com> http://dlbeer.co.nz/
PGP: BA6E 0B26 1F89 246C E3F3  C910 1E58 C43A 160A 553B

simion96, 2018-03-18:
Hi Daniel, 

Sorry for leaving this issue open for so long, i forgot about it. In the end I wasn't able to figure out the installation, and ended up getting a VM from a professional in the field with TinyOS + mspdebug on it. For this I can't answer too many questions on the installation, but regarding the above:

* The JTAG connections were ok, haven't changed anything since getting the VM and the connection works fine

* That's correct, I was using the on-board supply, at the time, but even when trying to supply power externally, it would have the same symptoms

* The interface was JTAG

* No board faults as far as I'm aware as everything works fine now

Sorry I couldn't be of more use to diagnose this. Also note that I've been using an old JTAG device (msp-fet430uif v1.4a) since I got this VM as it's what it's been tested with. When I was having issues with the above problem i was trying both the JTAGs separately. If it's any use to you whether the Olimex jtag works now I could give it a try at some point and hopefully reply faster than this time 😃 

Cheers

closed 2018-05-11 by simion96