dlbeer/mspdebug #43: MSP430FR2433 Launchpad (ez-FET) troubles

crosser, 2017-12-02:
Sorry to use this place for support request, please point to a better place if there is any.

I got this new MSP430FR2433 Launchpad that TI is now selling for promo price, and initially tried to communicate with it over mspdebug 0.22 with tilib driver (libmsp430 from slac460w). It pretended to work, saying successful erase and write but did nothing.
Then I realized that there are newer versions of mspdebug and built it from the master. Now I got two problems:

1. First I did not realize that there is a new `ezfet` driver and tried `tilib` as before. It wanted to update FET firmware, I ran with `--allow-fw-update`, it said
```
MSPDebug version 0.25 - debugging tool for MSP430 MCUs
Copyright (C) 2009-2017 Daniel Beer <dlbeer@gmail.com>
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Chip info database from MSP430.dll v3.3.1.4 Copyright (C) 2013 TI, Inc.

Using new (SLAC460L+) API
MSP430_GetNumberOfUsbIfs
MSP430_GetNameOfUsbIf
Found FET: HID_FET
MSP430_Initialize: HID_FET
FET firmware update is required.
Starting firmware update (this may take some time)...
tilib: MSP430_FET_FwUpdate: MSP-FET / eZ-FET recovery failed (error = 73)
tilib: device initialization failed
```
and now it changed USB ID to 2047:0203 and ezfet driver does not recognize it. Did I brick the FET? Is there any chance to revive it?

2. I have a second board that I did not brick (it shows as 2047:0013), but when I try to connect to it with ezfet, it says HAL exception:
```
$ ./mspdebug ezfet
MSPDebug version 0.25 - debugging tool for MSP430 MCUs
Copyright (C) 2009-2017 Daniel Beer <dlbeer@gmail.com>
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Chip info database from MSP430.dll v3.3.1.4 Copyright (C) 2013 TI, Inc.

Reset communications...
Version: 2.10.0.3, HW: 0xaaacaaac
Reset firmware...
hal_proto_execute: HAL exception: 0xffff
hal_proto_execute: fid: 0x51
```
Any advice for me? Thanks!

larsbrinkhoff, 2017-12-02:
I got one of those boards too, so I'll follow this closely.

crosser, 2017-12-02:
@larsbrinkhoff please update if you make your board work.
Plus additional information for context: https://e2e.ti.com/support/microcontrollers/msp430/f/166/t/646982

dlbeer, 2017-12-02:
On Sat, Dec 02, 2017 at 01:26:31PM +0000, Eugene Crosser wrote:
> Sorry to use this place for support request, please point to a better place if there is any.
>
> I got this new MSP430FR2433 Launchpad that TI is now selling for promo price, and initially tried to [...]

Hi Eugene,

You need to contact TI for help with the library.

The "ezfet" driver is very experimental -- I wouldn't recommend using
it.

Cheers,
Daniel

-- 
Daniel Beer <dlbeer@gmail.com> http://dlbeer.co.nz/
PGP: BA6E 0B26 1F89 246C E3F3  C910 1E58 C43A 160A 553B

crosser, 2017-12-03:
Thanks for the reply, @dlbeer !

- So is it true that tilib driver _should_ work with eZ-FET probe?
- I would like to note that mspdebug 0.22 did not insist on updating the firmware, and pretended to work (but did not), while 0.25 did not want to work without reflashing the firmware. In both cases the same libmsp430.so was used.

Would it be possible/feasible to try to prevent reflashing with "wrong" firmware?

dlbeer, 2017-12-03:
On Sun, Dec 03, 2017 at 06:59:11AM +0000, Eugene Crosser wrote:
> Thanks for the reply, @dlbeer !
> 
> - So is it true that tilib driver _should_ work with eZ-FET probe?

The tilib driver should work with libmsp430.so. Whether libmsp430.so
works with the eZ-FET is up to TI, but I would expect so.

> - I would like to note that mspdebug 0.22 did not insist on updating the firmware, and pretended to work (but did not), while 0.25 did not want to work without reflashing the firmware. In both cases the same libmsp430.so was used.

Version L of libmsp430.so has a different ABI from previous versions on
64-bit systems. mspdebug 0.22 was released before version L of the
library, and expects the original ABI, which leads to data transferred
between mspdebug and the library becoming corrupted. Versions 0.24
onwards support both ABIs.

> Would it be possible/feasible to try to prevent reflashing with "wrong" firmware?

You'd have to ask TI that -- the library doesn't provide much control
over firmware flashing.

Cheers,
Daniel

-- 
Daniel Beer <dlbeer@gmail.com> http://dlbeer.co.nz/
PGP: BA6E 0B26 1F89 246C E3F3  C910 1E58 C43A 160A 553B

crosser, 2017-12-03:
OK, thank you @dlbeer !
Now I have some basic information for my conversation with TI :)
I am closing this ticket.

closed 2017-12-03 by crosser

larsbrinkhoff, 2017-12-05:
@crosser I see you managed to update the firmware.  Does mspdebug work now?

crosser, 2017-12-05:
1. I installed [pre-release CCS 7.4](http://processors.wiki.ti.com/index.php/Code_Composer_Studio_Beta_Downloads), it works on recent Ubuntu. It comes with it's own copy of libmsp430.so. I was able to revive one of the boards from the CCS GUI, the other with MSP430Flasher (TI's own analog of mspdebug) using the same libmsp430.
2. mspdebug is able to communicate with the launchpad using the same libmsp430.so. Do not know if it would be able to restore the probe firmware or not, and not inclined to try ;)

sjlongland, 2018-12-08:
I just ran into the same issue, bought a MSP430FR2433 board to evaluate MSP430… and generally "have a play".  I had used them years ago, but the big turn-off for me was the requirement of the proprietary `gdbproxy`.  Hence I never used them for my own personal projects.

When I saw `mspdebug`, I thought, brilliant, either TI have shared some data to make this happen, or people have figured it out enough to make it work.  I bought the MSP-EXP430FR2433 to give it a shot.  I hit the same issue.

Now… I don't have `libmsp430.so` on my machine, so `tilib` doesn't work.  I'm really not in favour of requiring proprietary blobs, it was this reason I've avoided MSP430 for the last 10 years even though there's some nice chips in this family and it's where I began my embedded software work.

So, the queries:
1. Is there some way we can help improve the `ezfet` support in `mspdebug`?  I understand it's experimental, how can we as a user group, help make it better?
2. Is there a recommended JTAG or Spy-by-wire interface that you'd suggest in place of `ezfet`?
3. Does someone know of an alternate firmware for the MSP430F5528 MCU that implements the `ezfet` that might work better with `mspdebug`?

dlbeer, 2018-12-09:
On Fri, Dec 07, 2018 at 11:50:27PM -0800, Stuart Longland wrote:
> Now… I don't have `libmsp430.so` on my machine, so `tilib` doesn't
> work.  I'm really not in favour of requiring proprietary blobs, it was
> this reason I've avoided MSP430 for the last 10 years even though
> there's some nice chips in this family and it's where I began my
> embedded software work.

There's no proprietary blob -- the source for libmsp430.so is available
under a BSD-like licence. TI name the source packages following the
pattern slac460*.zip. The last release I looked at was slac460y.zip:

    https://dlbeer.co.nz/articles/slac460y/index.html

Cheers,
Daniel

-- 
Daniel Beer <dlbeer@gmail.com> http://dlbeer.co.nz/
PGP: BA6E 0B26 1F89 246C E3F3  C910 1E58 C43A 160A 553B