dlbeer/mspdebug #44: HAL exception 0xffff with ezfet on MSP-EXP430FR6989

alex-shpilkin, 2018-01-19:
As mentioned [elsewhere](https://www.pabigot.com/msp430/exp430fr5969-launchpad-first-experiences/#comment-39), the `ezfet` driver does not work with the MSP-EXP430FR6989:

```
MSPDebug version 0.25 - debugging tool for MSP430 MCUs
[snip]
Reset communications...
Version: 2.11.0.1, HW: 0xaaaaaaaa
Reset firmware...
hal_proto_execute: HAL exception: 0xffff
hal_proto_execute: fid: 0x51
```

This happens both with a pristine board and after updating the firmware using the `tilib` driver (with latest `libmsp430` 3.11.000.001). The `tilib` driver works if one does not try `ezfet` before (judging by the LEDs, this is because the board gets stuck in debug mode and `tilib` considers it to be already in use).

dlbeer, 2018-01-22:
On Fri, Jan 19, 2018 at 06:02:30PM +0000, Alexander Shpilkin wrote:
> As mentioned [elsewhere](https://www.pabigot.com/msp430/exp430fr5969-launchpad-first-experiences/#comment-39), the `ezfet` driver does not work with the MSP-EXP430FR6989:
>
> ```
> MSPDebug version 0.25 [...]

The ezfet driver is experimental and unfinished. You probably shouldn't
use it unless you're planning on trying to develop it further. Use the
tilib driver instead.

-- 
Daniel Beer <dlbeer@gmail.com> http://dlbeer.co.nz/
PGP: BA6E 0B26 1F89 246C E3F3  C910 1E58 C43A 160A 553B

f4grx, 2020-04-30:
Hello,
I am using a raspberry pi so no msp430 dll is available, I need to use the ezfet driver.

```
 ./mspdebug -s C6BC846F12001B00 ezfet
MSPDebug version 0.25 - debugging tool for MSP430 MCUs
Copyright (C) 2009-2017 Daniel Beer <dlbeer@gmail.com>
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Chip info database from MSP430.dll v3.15.0.1 Copyright (C) 2013 TI, Inc.

Reset communications...
Version: 2.4.0.20, HW: 0xaaaaaaaa
Reset firmware...
hal_proto_execute: HAL exception: 0xffff
hal_proto_execute: fid: 0x51
pi@iot:~/code/mspdebug $
```

Is there any chance that this board will be supported?
What remains to be done for it to work?
I can program, how can I help?
Thanks.

dlbeer, 2020-04-30:
On Thu, Apr 30, 2020 at 07:38:37AM -0700, f4grx wrote:
> I am using a raspberry pi so no msp430 dll is available, I need to use the ezfet driver.
>
> ```
>  ./mspdebug -s C6BC846F12001B00 ezfet
> MSPDebug version 0.25 - debugging tool for MSP430 MCUs
> Copyright [...]

The ezfet driver is just an experiment, and it's a long way off from
being usable. My suggestion would be to build libmsp430.so from source
for the Raspberry Pi and use the "tilib" driver.

Cheers,
Daniel

-- 
Daniel Beer <dlbeer@gmail.com> http://dlbeer.co.nz/
PGP: BA6E 0B26 1F89 246C E3F3  C910 1E58 C43A 160A 553B

f4grx, 2020-05-04:
I managed to recompile msp430.dll for a raspi with minimal changes to find hid-libusb and get everything working.

I also have a goodfet to test but I have not tried that yet.