dlbeer/mspdebug #48: Unable to connect with GPIO driver

dalmago, 2018-03-01:
I am trying to use the GPIO driver on a Raspberry Pi 3, but I always get the following message:

```
Starting JTAG
jtag_init: fuse is blown
JTAG_power on
JTAG_connct 
JTAG ID: 0x00
pif: unexpected JTAG ID: 0x00
jtag_set_instruction_fetch: failed
gpio: initialization failed
led green
```
I know the fuse is not blown because I can use the JTAG with the FET430UIF device.
Does anyone have any similar issue?

dalmago, 2018-03-01:
Also tried it on a Raspberry Zero W and got the same error.

dlbeer, 2018-03-01:
The JTAG driver is a little bit experimental, but the fact that you're getting an all-zero JTAG ID suggests that perhaps the JTAG pin connections aren't right. I'd double-check those and try again.

dalmago, 2018-03-02:
I checked the connections and plugged a logic analyzer. It seems that it's some issue with the initialization sequence. I got it working, but sometimes some commands fail. I'll submit updates soon.

dalmago, 2018-03-02:
I connected a MSP-FET430UIF device via JTAG interface on my MSP430F5510 and opened mspdebug. The start sequence on the **test** and **reset** pins are as following:

[image]

Then I compared with the init sequence using the GPIOs:
[image]

It didn't seem quite right to me, so [I changed the code a little bit](https://github.com/dalmago/mspdebug/commit/75e52cfbdda1bd30c857d22f9c7d1df42d3eb8bd) to match the TI's [SLAU320AC document](http://www.ti.com/lit/ug/slau320ac/slau320ac.pdf) (page 23, case 1b). I also added some sleep time just in case, because when the pins are set as outputs, they have a '0' level, causing a rising edge on **reset**, so I thought it's better to wait some time before resetting it again.

[image]

I'm not sure if it'll work on other families of devices, but mspdebug connects **almost** successful now:
```
gpio tms= 4
gpio tdi= 3
gpio tdo= 2
gpio tck= 17
gpio rst= 22
gpio tst= 27
Starting JTAG
JTAG_power on
JTAG_connect 
led green
JTAG ID: 0x91
device_probe_id: tlv_read failed
warning: device ID probe failed
```

dalmago, 2018-03-06:
Shouldn't [these](https://github.com/dlbeer/mspdebug/blob/master/drivers/jtdev_gpio.c#L200) be **GPIO_TCK**?

dlbeer, 2018-03-06:
On Tue, Mar 06, 2018 at 01:30:51PM +0000, Matheus wrote:
> Shouldn't [these](https://github.com/dlbeer/mspdebug/blob/master/drivers/jtdev_gpio.c#L200) be **GPIO_TCK**?

No, TCLK is something else -- during flash programming, the MSP430 needs
a clock to be supplied on TDI within a particular range (a few hundred
kHz). This is one of the things that makes JTAG via software bit-banging
difficult if you don't have precise timing control.

Cheers,
Daniel

-- 
Daniel Beer <dlbeer@gmail.com> http://dlbeer.co.nz/
PGP: BA6E 0B26 1F89 246C E3F3  C910 1E58 C43A 160A 553B

Graniyo, 2018-04-04:
Did you also get a 0xAAAA at TDO under fuse check? I'm struggling to get access with GPIO too, but I don't know if the fuse is really blown. Have you got it working? I'm going to edit the code tomorrow and test myself.

dalmago, 2018-04-04:
Found out that the JTAG commands are different for MSP430F5xx (which mine is) and MSP430F6xx families, so I'm using mspdebug just with the BSL driver.

codehero, 2022-08-25:
I am getting almost identical output for the MSP430FR5739....is the FR family supported by JTAG?

dalmago, 2022-08-26:
Sorry friend, it's been a while I don't deal with the MSP430 ucs, I'll leave this to be answered by someone who is more up to date!