dlbeer/mspdebug #59: simio console ignores newlines

hosewiejacke, 2018-06-26:
### Issue

When writing "a\nb\nc" to 0xff you get

    c

instead of

    a
    b
    c

### Steps to reproduce

Source:

	.org 0x0200

	print_a:
		mov.b #0x61, &0xff
		mov.b #0x0a, &0xff

	print_b:
		mov.b #0x62, &0xff
		mov.b #0x0a, &0xff

	print_c:
		mov.b #0x63, &0xff

	ret

	.org 0xfffe
	.dw print_a

Assemble: `naken_asm test.s43`

Run:

    mspdebug sim
    prog out.hex
    simio add console c
    step 0 ...
    simio info c

dlbeer, 2018-06-27:
It *should* be line-buffered, but perhaps this doesn't work. I don't have naken_asm installed on my laptop currently -- would you be able to paste the S43 here (or email it to me)?

hosewiejacke, 2018-06-27:
Sure, here it is:

    :10020000F2406100FF00F2400A00FF00F24062008D
    :10021000FF00F2400A00FF00F2406300FF0030419F
    :02FFFE000002FF
    :00000001FF

dlbeer, 2018-06-27:
This appears to work correctly -- the "Buffer:" line in the "info" output shows the contents of the current line buffer. Previously-flushed line buffers are sent to stdout.

hosewiejacke, 2018-06-28:
Sorry, I expected to see all lines in "simio info <console>". Closing this...

closed 2018-06-28 by hosewiejacke