dlbeer/mspdebug #74: Linux: Ti launchpad is not detecting

harielamadu, 2018-09-04:
HI,
Ti's MSP-EXP430G2 launchpad is not detecting in Ubuntu,and getting an error message while running the "dmesg" command.I am using the mspdebug 0.25 for programming the board?
[image]
[image]
[image]

dlbeer, 2018-09-04:
On Tue, Sep 04, 2018 at 08:33:49AM +0000, harielamadu wrote:
> HI,
> Ti's MSP-EXP430G2 launchpad is not detecting in Ubuntu,and getting an error message while running the "dmesg" command.I am using the mspdebug 0.25 for programming the board?
> [image]
> [image]
> [image]

Looks like a bad connection or some kind of electrical fault. Have you
double-checked the USB cable and connectors?

-- 
Daniel Beer <dlbeer@gmail.com> http://dlbeer.co.nz/
PGP: BA6E 0B26 1F89 246C E3F3  C910 1E58 C43A 160A 553B

kianmeng, 2018-09-05:
@harielamadu It [seems to work for me](http://www.kianmeng.org/2017/02/msp430-setting-up-launchpad-development.html) but that was in Ubuntu 16.10.