dlbeer/mspdebug #75: Is MSPDEBUG_TILIB_PATH broken on Windows?

melak, 2018-09-15:
https://github.com/dlbeer/mspdebug/blob/457c0c3f00c3791289275093cc7d894bc0ebd010/drivers/tilib_api.c#L768-L771

Does this work on Windows?

dlbeer, 2018-09-15:
On Sat, Sep 15, 2018 at 01:03:16PM -0700, melak wrote:
> https://github.com/dlbeer/mspdebug/blob/457c0c3f00c3791289275093cc7d894bc0ebd010/drivers/tilib_api.c#L768-L771
> 
> Does this work on Windows?

I haven't tried it, but Windows does accept '/' as a path separator, so
I would expect it to work.

-- 
Daniel Beer <dlbeer@gmail.com> http://dlbeer.co.nz/
PGP: BA6E 0B26 1F89 246C E3F3  C910 1E58 C43A 160A 553B

DanielO, 2018-09-16:
Hmm, to be honest I assumed it would be compiled with something like cygwin so it would work but I haven't tested it.
I don't have the tools to build it for Windows but I do have a Windows 10 VM I could probably test it with..

melak, 2018-09-16:
OK, just checking.

closed 2018-09-16 by melak