dlbeer/mspdebug #77: Mac

Euler1707, 2018-10-29:
On Mac Os High Sierra 10.13.6 (17G65)

MSPDebug version 0.25 - debugging tool for MSP430 MCUs
Copyright (C) 2009-2017 Daniel Beer <dlbeer@gmail.com>
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Chip info database from MSP430.dll v3.3.1.4 Copyright (C) 2013 TI, Inc.

rf2500: failed to open RF2500 device

$ mspdebug --usb-list

Devices on bus 020:
    020:010 0451:f432 eZ430-RF2500 [serial: 13FF55ADB2161B22]
    020:009 05ac:0259

dlbeer, 2018-10-29:
Is this the 0.25 release you're using, or the latest git master? If it's the release, could you retry with the git master branch?

Euler1707, 2018-10-29:
You are right I apologize, it works after building from source. I was originally using the Homebrew release.

glingy, 2020-06-06:
I can confirm that for me on macOS Catalina, I get no error in 0.23 or 0.24, that same error in 0.25, and again no error in the current master, so it seems master fixed the problem.