dlbeer/mspdebug #78: gdb and mspdebug communication

jlbirccyn, 2018-11-06:
Hello,

mspdebug 0.24
msp430-gdb 7.2
Mac OS X 10.11.6

I am trying to debug using gdb and mspdebug. I am using mspdebug 0.24 that comes in Energia distribution and msp430-gdb from Energia too.

```
./mspdebug tilib gdb
```

leads to

```
MSPDebug version 0.24 - debugging tool for MSP430 MCUs
Copyright (C) 2009-2016 Daniel Beer <dlbeer@gmail.com>
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Chip info database from MSP430.dll v3.3.1.4 Copyright (C) 2013 TI, Inc.

Using new (SLAC460L+) API
MSP430_GetNumberOfUsbIfs
MSP430_GetNameOfUsbIf
Found FET: usbmodem1421
MSP430_Initialize: usbmodem1421
Firmware version is 31300001
MSP430_VCC: 3000 mV
MSP430_OpenDevice
MSP430_GetFoundDevice
Device: MSP430FR6989 (id = 0x0160)
3 breakpoints available
MSP430_EEM_Init
Chip ID data:
  ver_id:         81a8
  ver_sub_id:     0000
  revision:       21
  fab:            55
  self:           5555
  config:         20
Device: MSP430FR6989 [FRAM]
Bound to port 2000. Now waiting for connection...
```

I launch gdb from another terminal and

```
(gdb) tar extended-remote :2000
Remote debugging using :2000
Remote register badly formatted: T0500:00440000;01:f4230000;02:00000000;03:00000000;04:10080000;05:0c5a0000;06:10080000;07:5aa50000;08:93000000;09:00000000;0a:08d40000;0b:45010000;0c:89510000;0d:00000000;0e:e0530000;0f:00000000;
here: 0000;01:f4230000;02:00000000;03:00000000;04:10080000;05:0c5a0000;06:10080000;07:5aa50000;08:93000000;09:00000000;0a:08d40000;0b:45010000;0c:89510000;0d:00000000;0e:e0530000;0f:00000000;
```

Best regards

dlbeer, 2018-11-06:
On Tue, Nov 06, 2018 at 02:34:47PM +0000, Jean-Luc Béchennec wrote:
> Hello,
>
> mspdebug 0.24
> msp430-gdb 7.2
> Mac OS X 10.11.6
>
> I am trying to debug using gdb and mspdebug. I am using mspdebug 0.24 that comes in Energia distribution and msp430-gdb from Energia too.
>
> ``` [...]

Would you be able to try compiling the latest mspdebug from source with:

    make CFLAGS=-DDEBUG_GDB

...and then try this binary with gdb? It should emit some extra output
which may be useful.

Cheers,
Daniel

-- 
Daniel Beer <dlbeer@gmail.com> http://dlbeer.co.nz/
PGP: BA6E 0B26 1F89 246C E3F3  C910 1E58 C43A 160A 553B

jlbirccyn, 2018-11-07:
Hello

Thanks to take time

Here is the output:

```
(mspdebug) gdb
Bound to port 2000. Now waiting for connection...
Client connected from 127.0.0.1:62467
Clearing all breakpoints...
starting GDB reader loop...
<- $qSupported:multiprocess+;qRelocInsn+#2a
process_gdb_command: qSupported:multiprocess+;qRelocInsn+
-> $PacketSize=4000#f4
<- $!#21
process_gdb_command: !
process_gdb_command: unknown command !
-> $#00
<- $Hg0#df
process_gdb_command: Hg0
process_gdb_command: unknown command Hg0
-> $#00
<- $?#3f
process_gdb_command: ?
-> $T0500:00440000;01:f4230000;02:00000000;03:00000000;04:10080000;05:0c5a0000;06:10080000;07:5aa50000;08:32000000;09:00000000;0a:f08a0000;0b:ba000000;0c:b92e0000;0d:00000000;0e:f85c0000;0f:00000000;#58
<- $Hc-1#09
process_gdb_command: Hc-1
process_gdb_command: unknown command Hc-1
-> $#00
<- $qC#b4
process_gdb_command: qC
process_gdb_command: unknown command qC
-> $#00
<- $qAttached#8f
process_gdb_command: qAttached
process_gdb_command: unknown command qAttached
-> $#00
Connection closed
... reader loop returned
```

Best regards

dlbeer, 2018-11-08:
On Wed, Nov 07, 2018 at 07:22:50AM -0800, Jean-Luc Béchennec wrote:
> Hello
> 
> Thanks to take time
> 
> Here is the output:

Thanks for that. I'll build msp430-elf-gcc here and see if I can
reproduce and fix the problem.

-- 
Daniel Beer <dlbeer@gmail.com> http://dlbeer.co.nz/
PGP: BA6E 0B26 1F89 246C E3F3  C910 1E58 C43A 160A 553B

dlbeer, 2018-11-09:
This seems to work with the latest gdb (8.2) compiled for target
msp430-elf, so I'm guessing Energia bundles an old gdb which we're not
detecting correctly.

If you find and comment out these lines in ui/gdb.c:

		if (strstr(buf, "multiprocess+"))
			register_bytes = 4;

...does Energia's gdb then work with the recompiled mspdebug?

-- 
Daniel Beer <dlbeer@gmail.com> http://dlbeer.co.nz/
PGP: BA6E 0B26 1F89 246C E3F3  C910 1E58 C43A 160A 553B

glingy, 2020-06-06:
This is a bit old, but I had the same problem and using msp430-elf-gcc, it works, but Energia's didn't. With those two lines commented out, it seems to work great.

maxgerhardt, 2020-06-06:
-- this bug seems to be preventing the PlatformIO integration for msp430 to be working correctly with gdb (https://community.platformio.org/t/badly-formatted-error-during-debug/13916). Is there a way the program can be patched so it's compatible with newer and older gdb versions?

michaelde, 2020-07-11:
> 
> 
> -- this bug seems to be preventing the PlatformIO integration for msp430 to be working correctly with gdb >(https://community.platformio.org/t/badly-formatted-error-during-debug/13916). Is there a way the >program can be patched so it's compatible with newer and older gdb versions?

**Thanks for Your hint! Don't know either how to implement this patch for
PlatformIO running on a Windows PC.**