dlbeer/mspdebug #81: support for Launchpad EXP430GET

dlarue, 2019-01-10:
I was borrowing an EXP430G2 to prog some msp4302553 boards and went to get one from Ti only to find out Ti no longer sells them. They sell the EXP430G2ET in its place but mspdebug doesn't see it.

As a pointer that a Linux tool does work:
In my searches for a solution I found Energia IDE and found the latest version( 21/1.8.7E21 ) does work with the G2ET. Enabling verbose output showed it using DSLite to do the work.  I don't know what Ti changed to the programmer interface between G2 and G2ET but maybe the differences can be gleaned from the dslite project.

dlbeer, 2019-01-10:
On Wed, Jan 09, 2019 at 08:53:08PM -0800, Doug LaRue wrote:
> I was borrowing an EXP430G2 to prog some msp4302553 boards and went to get one from Ti only to find out Ti no longer sells them. They sell the EXP430G2ET in its place but mspdebug doesn't see it. [...]

Hi Doug,

Did you try the TI library with mspdebug's "tilib" driver, or something
else? I'd expect that libmsp430.so should work, since it's supposed to
be able to work with every debugger TI produces.

Cheers,
Daniel

-- 
Daniel Beer <dlbeer@gmail.com> http://dlbeer.co.nz/
PGP: BA6E 0B26 1F89 246C E3F3  C910 1E58 C43A 160A 553B

chrissphinx, 2019-05-11:
I can't get this to work either:

```
$ lsusb
Bus 001 Device 002: ID 0e0f:000b VMware, Inc. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 007: ID 2047:0013 Texas Instruments 
Bus 002 Device 005: ID 0e0f:0008 VMware, Inc. 
Bus 002 Device 003: ID 0e0f:0002 VMware, Inc. Virtual USB Hub
Bus 002 Device 002: ID 0e0f:0003 VMware, Inc. Virtual Mouse
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub

$ mspdebug tilib
MSPDebug version 0.25 - debugging tool for MSP430 MCUs
Copyright (C) 2009-2017 Daniel Beer <dlbeer@gmail.com>
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Chip info database from MSP430.dll v3.13.0.601 Copyright (C) 2013 TI, Inc.

Using new (SLAC460L+) API
MSP430_GetNumberOfUsbIfs
MSP430_GetNameOfUsbIf
No unused FET found.
```

This is the new MSP-EXP430G2ET Launchpad using the provided MSP430G2553 chip. I followed [the instructions here](https://dlbeer.co.nz/articles/slac460y/index.html) to build `libmsp430.so`.

I also tried `ezfet` driver since I read that is what is used on this board:

```
$ mspdebug ezfet -U 002:005
MSPDebug version 0.25 - debugging tool for MSP430 MCUs
Copyright (C) 2009-2017 Daniel Beer <dlbeer@gmail.com>
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Chip info database from MSP430.dll v3.13.0.601 Copyright (C) 2013 TI, Inc.

transport/cdc_acm.c: can't claim interface: Operation not permitted
transport/cdc_acm.c: failed to open interface
fet3: failed to open transport
$ sudo !!
sudo mspdebug ezfet -U 002:005
[sudo] password for sphinx: 
MSPDebug version 0.25 - debugging tool for MSP430 MCUs
Copyright (C) 2009-2017 Daniel Beer <dlbeer@gmail.com>
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Chip info database from MSP430.dll v3.13.0.601 Copyright (C) 2013 TI, Inc.

Reset communications...
Version: 2.12.0.0, HW: 0xaaadaaad
Reset firmware...
hal_proto_execute: HAL exception: 0xffff
hal_proto_execute: fid: 0x51
```

Weirdly I had to use `sudo` despite being in the `dialout` group (and being able to successfully open the device using `minicom`). Any way forward with these new boards?

dlarue, 2019-05-11:
I found out that CCSv8 worked and there was even an installation of mspdebug there.
/ccsv8/ccs_base/DebugServer/bin/mspdebug

chrissphinx, 2019-05-11:
@dlarue but doesn't CCS use DSLite for debug/programming the board?

dlarue, 2019-05-12:
@chrissphinx It was a while ago when I want down the path trying to get mspdebug to work with the new EXp430GET...  All I can say at this point is that CCS v8 lets me compile, upload and debug using the EXP430GET.    I just tested if the version of mspdebug in the CCSv8 directory(posted above) works with the EXP430GET and it does not.  I know CCSv8 is a large app but memory and disk space is so cheap these days...

Have you tried adding the USB prod/dev IDs to the mspdebug source and see if it works?

dlarue, 2019-05-12:
@dlbeer @chrissphinx   using "mspdebug tilib" with the libmsp430.so worked!

I got the libmsp430.so from:
./Downloads/energia-1.8.7E21/hardware/tools/DSLite/DebugServer/drivers/libmsp430.so
and one from:
./Development/ti/ccsv8/ccs_base/DebugServer/drivers/libmsp430.so
and both worked.

I compiled in CCSv8 and loaded Debug/Blink.out changing the blink rate each time and validating the FET430 device LED blinked faster or slower.

chrissphinx, 2019-05-12:
@dlarue Are you using another device as opposed to simply plugging the G2ET board in via USB? I'm reading in the manpage that `tilib` driver will not support a USB connection?

I'm using current master of `mspdebug` and pulled the library out from CCS v9.0.1 but still just get:
```
MSPDebug version 0.25 - debugging tool for MSP430 MCUs
Copyright (C) 2009-2017 Daniel Beer <dlbeer@gmail.com>
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Chip info database from MSP430.dll v3.13.0.601 Copyright (C) 2013 TI, Inc.

Using new (SLAC460L+) API
MSP430_GetNumberOfUsbIfs
MSP430_GetNameOfUsbIf
No unused FET found.
```

dlarue, 2019-05-12:
@chrissphinx   it is the EXP430G2ET from TI.  Here my dmesg output:

[ 3602.392666] usb 1-1: new full-speed USB device number 11 using xhci_hcd
[ 3602.543040] usb 1-1: New USB device found, idVendor=2047, idProduct=0013
[ 3602.543041] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 3602.543043] usb 1-1: Product: MSP Tools Driver
[ 3602.543044] usb 1-1: Manufacturer: Texas Instruments
[ 3602.543045] usb 1-1: SerialNumber: AE33816E04001100
[ 3602.544073] cdc_acm 1-1:1.0: ttyACM0: USB ACM device
[ 3602.544683] cdc_acm 1-1:1.2: ttyACM1: USB ACM device

chrissphinx, 2019-05-12:
Here is my own:
```
[  698.459135] usb 2-2.2: USB disconnect, device number 5
[  701.984053] usb 2-2.2: new full-speed USB device number 6 using uhci_hcd
[  702.107645] usb 2-2.2: New USB device found, idVendor=2047, idProduct=0013
[  702.107648] usb 2-2.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  702.107650] usb 2-2.2: Product: MSP Tools Driver
[  702.107652] usb 2-2.2: Manufacturer: Texas Instruments
[  702.107653] usb 2-2.2: SerialNumber: A79234510A000D00
[  702.921384] cdc_acm 2-2.2:1.0: ttyACM0: USB ACM device
[  702.923023] cdc_acm 2-2.2:1.2: ttyACM1: USB ACM device
```

What version of linux are you using? The only difference I can see is that the `cdc_acm` driver I have is newer.

dlarue, 2019-05-12:
I'm on Kubuntu 18.04

dlarue, 2019-05-12:
you know it's going to be one of those palm-smacks-forehead type of problems right? :-)

dlbeer, 2019-05-12:
chrissphinx <notifications@github.com> writes:
> Weirdly I had to use `sudo` despite being in the `dialout` group (and
> being able to successfully open the device using `minicom`). Any way
> forward with these new boards?

Hi Chris,

When you say you tried sudo, do you mean that you tried it with the
tilib driver, or just the ezfet attempt?

Have you tried with the version of libmsp430.so that Doug suggested
(sorry if you have and I missed it, I'm following this via email).

Cheers,
Daniel

--
Daniel Beer <dlbeer@gmail.com> http://dlbeer.co.nz/
PGP: BA6E 0B26 1F89 246C E3F3  C910 1E58 C43A 160A 553B

dlarue, 2019-05-12:
my udev rules file goes like this:
#TI MSP430UIF
ATTRS{idVendor}=="2047",ATTRS{idProduct}=="0010",MODE="0666"
ATTRS{idVendor}=="2047",ATTRS{idProduct}=="0013",MODE="0666"
ATTRS{idVendor}=="2047",ATTRS{idProduct}=="0014",MODE="0666"
ATTRS{idVendor}=="2047",ATTRS{idProduct}=="0203",MODE="0666"
ATTRS{idVendor}=="2047",ATTRS{idProduct}=="0204",MODE="0666"
ATTRS{idVendor}=="0451",ATTRS{idProduct}=="f432",MODE="0666"

But wait, there's more!  I just realized I had another rule with the 2047/0013 ven/prod ids in it.
I called it 70-mm-no-ti-emulators.rules and it goes like this:
ACTION!="add|change", GOTO="mm_usb_device_blacklist_end"
SUBSYSTEM!="usb", GOTO="mm_usb_device_blacklist_end"
ENV{DEVTYPE}!="usb_device",  GOTO="mm_usb_device_blacklist_end"

# TI USB Emulators
ATTRS{idVendor}=="2047", ATTRS{idProduct}=="0010", ENV{ID_MM_DEVICE_IGNORE}="1"
ATTRS{idVendor}=="2047", ATTRS{idProduct}=="0013", ENV{ID_MM_DEVICE_IGNORE}="1"
ATTRS{idVendor}=="2047", ATTRS{idProduct}=="0014", ENV{ID_MM_DEVICE_IGNORE}="1"
ATTRS{idVendor}=="2047", ATTRS{idProduct}=="0203", ENV{ID_MM_DEVICE_IGNORE}="1"
ATTRS{idVendor}=="2047", ATTRS{idProduct}=="0204", ENV{ID_MM_DEVICE_IGNORE}="1"
ATTRS{idVendor}=="0451", ATTRS{idProduct}=="F432", ENV{ID_MM_DEVICE_IGNORE}="1"

LABEL="mm_usb_device_blacklist_end"

chrissphinx, 2019-05-13:
@dlbeer Yes, I am using the `libmsp430.so` from CCS v9.0.1, so I'm not sure if that's any different but the line `Chip info database from MSP430.dll v3.13.0.601 Copyright (C) 2013 TI, Inc.` seems to suggest we have the same `.so`. I only used `sudo` when I was trying out `ezfet`.

I am using Ubuntu 18.04 as well, but through an emulator. I also cannot get CCS to see the board from inside the VM, but it works fine on my host machine (macOS).

dlbeer, 2019-05-13:
chrissphinx <notifications@github.com> writes:
> @dlbeer Yes, I am using the `libmsp430.so` from CCS v9.0.1, so I'm not
> sure if that's any different but the line `Chip info database from
> MSP430.dll v3.13.0.601 Copyright (C) 2013 TI, Inc.` seems to suggest
> we have the same `.so`. I only used `sudo` when I was trying out
> `ezfet`.

That line is from information extracted separately and bundled with
mspdebug -- not the .so.

I'd suggest trying the tilib driver with sudo. The error may be due to
insufficient permissions, which the TI library will just report as the
lack of an available device.

Cheers,
Daniel

--
Daniel Beer <dlbeer@gmail.com> http://dlbeer.co.nz/
PGP: BA6E 0B26 1F89 246C E3F3  C910 1E58 C43A 160A 553B

chrissphinx, 2019-05-17:
@dlbeer the issue ended up being this: http://e2e.ti.com/support/microcontrollers/msp430/f/166/t/713463?CCS-MSP-FET430UIF-MSP-FET430UIF-is-already-in-use-on-Ubuntu-18-04

But, @dlarue you are reporting that removing all the udev rules still allows you to make a connection?

After installing the `libmsp430.so` into the correct location, it seems to be talking to the new board just fine. Thank you for all the help on this!

As an aside, does anyone know the legalities behind the `libmsp430.so` library or even the supposedly "open source" `msp430-elf-gcc`? To obtain them both you seem to need a TI account and that is a real hassle for my students, am I allowed to distribute a package that will install everything they need if this library and tool are included?

dlarue, 2019-05-17:
@chrissphinx   if you are teaching students I can't stress enough the benefits of making one bootable ISO with all the software, documentation, samples etc pre-configured on the bootable ISO.  I've been down the road a few times dealing with so many different configuration and the time wasted on setup.  Now I spend the time setting up a custom system using Customizer-gui and let students download the ISO, put it on a USB thumb drive and verify it boots before classes even start.

And yes, I moved the ti specific udev rules files out of their location and rebooted before being surprised that mspdebug still recognized/found the FET.

chrissphinx, 2019-05-17:
@dlarue yes, I'm exploring doing something to that effect, though in general I don't like serving .ISOs since they are quite large.

Can I ask, although we're getting off topic here, are you using `msp430-gcc` or the new `msp430-elf-gcc`? My understanding is that `msp430-gcc` is no longer supported and my guess is that it will be removed from Ubuntu repos at some point, so I'm wondering whether to switch now or keep using the old stuff until I'm forced to move...

dlarue, 2019-05-17:
I serve the ISO files from a google drive so not bandwidth issues.

For msp430 I have CCS v8 running and works fine.  Finally full Linux support or at least functionality.

dlbeer, 2019-05-18:
chrissphinx <notifications@github.com> writes:
> As an aside, does anyone know the legalities behind the `libmsp430.so`
> library or even the supposedly "open source" `msp430-elf-gcc`? To
> obtain them both you seem to need a TI account and that is a [...]

Hi Chris,

libmsp430.so is BSD-licenced (source code is available on the TI
website). GCC and derivatives including msp430-elf-gcc are GPL. You're
free to distribute both to your students.

Cheers,
Daniel

--
Daniel Beer <dlbeer@gmail.com> http://dlbeer.co.nz/
PGP: BA6E 0B26 1F89 246C E3F3  C910 1E58 C43A 160A 553B

chrissphinx, 2019-05-18:
@dlbeer Thank you very much for this insight, Daniel!

Brad-R, 2020-07-13:
For what it's worth, I encountered a similar problem this week, that required a slightly different solution.
1. Remove mspdebug v0.22 and install mspdebug 0.25.
2. Enable 'tilib' option by installing libmsp430.so.  Building from source failed, so I copied libmsp430.so from CCS v9.0 to the /usr/lib/x86_64-linux-gnu/ directory.
That ran, but gave the message "No unused FET found."
3. Install TI's new FET drivers from http://software-dl.ti.com/msp430/msp430_public_sw/mcu/msp430/MSP430_FET_Drivers/latest/index_FDS.html
That reported "FET firmware update is required.  Re-run with --allow-fw-update to perform a firmware update."
4. mspdebug tilib --allow-fw-update
After that step, mspdebug is working and talking to my MSP-EXP430G2ET Launchpad.

chrissphinx, 2020-12-05:
I will just comment here to echo @Brad-R 's approach is what we are currently doing, I am bundling `libmsp430.so` (from a CCS install on Linux) with an install script that currently seems to work with both 18.04 and 20.04 Ubuntu. Have not yet had to install FET drivers, but the `--allow-fw-update` is frequently required. @Brad-R, what version of Ubuntu was this under?

Brad-R, 2020-12-05:
Not Ubuntu; I'm using Debian 10 "Buster", 64-bit.

dlarue, 2020-12-05:
That would mean Ubuntu 18.04 - 19.10 according to this list:
Ubuntu            Debian  
20.04  focal      bullseye/ sid   - 11
19.10  eoan       buster  / sid   - 10
19.04  disco      buster  / sid
18.10  cosmic     buster  / sid
18.04  bionic     buster  / sid
17.10  artful     stretch / sid   - 9
17.04  zesty      stretch / sid
16.10  yakkety    stretch / sid
16.04  xenial     stretch / sid
15.10  wily       jessie  / sid   - 8
15.04  vivid      jessie  / sid
14.10  utopic     jessie  / sid
14.04  trusty     jessie  / sid
13.10  saucy      wheezy  / sid   - 7
13.04  raring     wheezy  / sid
12.10  quantal    wheezy  / sid
12.04  precise    wheezy  / sid
11.10  oneiric    wheezy  / sid
11.04  natty      squeeze / sid   - 6
10.10  maverick   squeeze / sid
10.04  lucid      squeeze / sid