dlbeer/mspdebug #83: Simulating UART

chrissphinx, 2019-04-24:
We are using [this library by Stefan Wendler](https://github.com/wendlers/libemb) to give students an easy way to have a "command line like" experience when programming the MSP430.

However, for automated testing this poses a problem. It seems `mspdebug sim` can handle timers, gpio and interrupts caused by those pins no problem, but is there a way to simulate the UART interface so I would be able to connect to it using a program like `minicom` just as if it is programmed onto the actual board?

dlbeer, 2019-04-24:
chrissphinx <notifications@github.com> writes:
> We are using [this library by Stefan
> Wendler](https://github.com/wendlers/libemb) to give students an easy
> way to have a "command line like" experience when programming the
> MSP430.
>
> However, for [...]

The closest thing that currently exists is the "console" simio module,
but this will only simulate transmission. I take it this isn't enough
for what you're doing?

What you're describing could certainly be implemented as a new simio
module though -- you could allocate a pseudo-tty under Linux or BSD and
translate register access to reads and writes to the tty. Then you could
spin up minicom or similar and talk to the pseudo-tty.

Cheers,
Daniel

--
Daniel Beer <dlbeer@gmail.com> http://dlbeer.co.nz/
PGP: BA6E 0B26 1F89 246C E3F3  C910 1E58 C43A 160A 553B

tgtakaoka, 2019-04-24:
Having UART simio sounds great!

chrissphinx, 2019-04-30:
@dlbeer thanks for the reply, Daniel, are you saying it will only simulate transmission from the msp430 to the host machine or from the host to the msp430?

dlbeer, 2019-04-30:
chrissphinx <notifications@github.com> writes:
>    [1]@dlbeer thanks for the reply, Daniel, are you saying it will only
>    simulate transmission from the msp430 to the host machine or from the host
>    to the msp430?

Currently it only simulates transmission from the MSP430 to the host
machine, and in a fairly simple way -- there's no virtual TTY or
anything like that.

-- 
Daniel Beer <dlbeer@gmail.com> http://dlbeer.co.nz/
PGP: BA6E 0B26 1F89 246C E3F3  C910 1E58 C43A 160A 553B

chrissphinx, 2019-04-30:
Actually I think that would work for my needs because I could put a driver in front of code I want to test the simply sends characters to the host.

I assume that means I need to create a `console` peripheral that is "listening" at `UCA0TXBUF` (or `0x0067` for the MSP430G2553)?

dlbeer, 2019-04-30:
chrissphinx <notifications@github.com> writes:
>    I assume that means I need to create a console peripheral that is
>    "listening" at UCA0TXBUF (or 0x0067 for the MSP430G2553)?

Yes, that's correct. Any byte writes to the configured address will be
printed on stdout.

Cheers,
Daniel

--
Daniel Beer <dlbeer@gmail.com> http://dlbeer.co.nz/
PGP: BA6E 0B26 1F89 246C E3F3  C910 1E58 C43A 160A 553B

chrissphinx, 2019-04-30:
Sorry, Daniel, I had edited my previous reply. But, I actually got this to function by forcibly writing to `IFG` special function register:

```
(mspdebug) md 0x0003 1
    00003: 00                                              |.               |
(mspdebug) mw 0x0003 FF
(mspdebug) run
Running. Press Ctrl+C to interrupt...
Hello, World!
Hello, World!
```

Then no need to write to it after that? Any idea what's going on with this special function register in the simulator?

dlbeer, 2019-05-01:
chrissphinx <notifications@github.com> writes:
> Sorry, Daniel, I had edited my previous reply. But, I actually got this to function by forcibly writing to `IFG` special function register:
>
> ```
> (mspdebug) md 0x0003 1
>     00003: [...]

It may be that your code is waiting for the TX ready flag to be set for
the UART peripheral. That's not simulated, but manually setting the SFR
byte to 0xff will make it appear set to the code if that's the case.

-- 
Daniel Beer <dlbeer@gmail.com> http://dlbeer.co.nz/
PGP: BA6E 0B26 1F89 246C E3F3  C910 1E58 C43A 160A 553B

chrissphinx, 2019-05-01:
Okay great, I think I have all that I need for the moment, then. Thank you for the help, Daniel!

closed 2019-05-01 by chrissphinx