dlbeer/mspdebug #84: Write failure error code 7

bommuuu, 2019-06-08:
Writing failure to the msp4305152.
Error code 7:
Error reading data in line number xx.

The same hex file successfully working on other mechines.
Os: ubuntu
Hardware:raspberry pi b+

dlbeer, 2019-06-09:
Shravan Bommu <notifications@github.com> writes:
> Writing failure to the msp4305152.
> Error code 7:
> Error reading data in line number xx.
>
> The same hex file successfully working on other mechines.
> Os: ubuntu
> Hardware:raspberry pi b+

Hi Shravan,

Are you able to post a full transcript of the mspdebug session and
perhaps the hex file as well?

Cheers,
Daniel

--
Daniel Beer <dlbeer@gmail.com> http://dlbeer.co.nz/
PGP: BA6E 0B26 1F89 246C E3F3  C910 1E58 C43A 160A 553B

bommuuu, 2019-06-10:
Hi Danie,
Please find the full log below, Sorry i dont have permissions to share the hex files.
Regards,
Shravan
11:11:08 Started by upstream project "34000-134_Soteria_VID_Slice_FW_CI_CD" build number 89
11:11:08 originally caused by:
11:11:08  Started by user Shravan Bommu
11:11:08 Building on master in workspace /var/lib/jenkins/workspace/34000-134_Soteria_VID_Slice_Load
11:11:08 [34000-134_Soteria_VID_Slice_Load] $ /bin/sh -xe /tmp/jenkins818564310970748643.sh
11:11:08 + PROD_NAME=34000-134
11:11:08 + BUILD_WORKSPACE=34000-134_Soteria_VID_Slice_FW_CI_CD
11:11:08 + FILENAME=34000-134.txt
11:11:08 + echo FW Filename is 34000-134.txt
11:11:08 FW Filename is 34000-134.txt
11:11:08 + echo Jenkins Workspace is /var/lib/jenkins/workspace/34000-134_Soteria_VID_Slice_Load
11:11:08 Jenkins Workspace is /var/lib/jenkins/workspace/34000-134_Soteria_VID_Slice_Load
11:11:08 + pwd
11:11:08 /var/lib/jenkins/workspace/34000-134_Soteria_VID_Slice_Load
11:11:08 + cp ../34000-134_Soteria_VID_Slice_FW_CI_CD/svn_src/34000-134.txt 34000-134.txt
11:11:08 + echo Current directory is...
11:11:08 Current directory is...
11:11:08 + pwd
11:11:08 /var/lib/jenkins/workspace/34000-134_Soteria_VID_Slice_Load
11:11:08 + echo Files present are...
11:11:08 Files present are...
11:11:08 + ls -lrt
11:11:08 total 52
11:11:08 -rw-r--r-- 1 jenkins jenkins 47550 Jun  6 11:11 34000-134.txt
11:11:08 SSH: Connecting from host [v-af-jenkins1]
11:11:08 SSH: Connecting with configuration [test_stand_7] ...
11:11:12 SSH: EXEC: STDOUT/STDERR from command [mspdebug tilib --allow-fw-update erase
11:11:12 mspdebug tilib 'load 
11:11:12 /home/pi/tmp/34000-134.txt'] ...
11:11:13 MSPDebug version 0.25 - debugging tool for MSP430 MCUs
11:11:13 Copyright (C) 2009-2017 Daniel Beer <dlbeer@gmail.com>
11:11:13 This is free software; see the source for copying conditions.  There is NO
11:11:13 warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
11:11:13 Chip info database from MSP430.dll v3.3.1.4 Copyright (C) 2013 TI, Inc.
11:11:13 
11:11:13 Using new (SLAC460L+) API
11:11:13 MSP430_GetNumberOfUsbIfs
11:11:13 MSP430_GetNameOfUsbIf
11:11:13 Found FET: ttyACM0
11:11:13 MSP430_Initialize: ttyACM0
11:11:13 Firmware version is 31300601
11:11:13 MSP430_VCC: 3000 mV
11:11:17 MSP430_OpenDevice
11:11:17 MSP430_GetFoundDevice
11:11:17 Device: MSP430F5152 (id = 0x00ac)
11:11:17 3 breakpoints available
11:11:17 MSP430_EEM_Init
11:11:18 Chip ID data:
11:11:18   ver_id:         802c
11:11:18   ver_sub_id:     0000
11:11:18   revision:       30
11:11:18   fab:            55
11:11:18   self:           5555
11:11:18   config:         10
11:11:18   fuses:          55
11:11:18 Device: MSP430F5152
11:11:18 Erasing...
11:11:18 MSP430_Run
11:11:19 MSP430_Close
11:11:19 MSPDebug version 0.25 - debugging tool for MSP430 MCUs
11:11:19 Copyright (C) 2009-2017 Daniel Beer <dlbeer@gmail.com>
11:11:19 This is free software; see the source for copying conditions.  There is NO
11:11:19 warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
11:11:19 Chip info database from MSP430.dll v3.3.1.4 Copyright (C) 2013 TI, Inc.
11:11:19 
11:11:19 Using new (SLAC460L+) API
11:11:19 MSP430_GetNumberOfUsbIfs
11:11:19 MSP430_GetNameOfUsbIf
11:11:19 Found FET: ttyACM0
11:11:19 MSP430_Initialize: ttyACM0
11:11:19 Firmware version is 31300601
11:11:19 MSP430_VCC: 3000 mV
11:11:23 MSP430_OpenDevice
11:11:24 MSP430_GetFoundDevice
11:11:24 Device: MSP430F5152 (id = 0x00ac)
11:11:24 3 breakpoints available
11:11:24 MSP430_EEM_Init
11:11:24 Chip ID data:
11:11:24   ver_id:         802c
11:11:24   ver_sub_id:     0000
11:11:24   revision:       30
11:11:24   fab:            55
11:11:24   self:           5555
11:11:24   config:         10
11:11:24   fuses:          55
11:11:24 Device: MSP430F5152
11:11:24 Writing  128 bytes at 1800...
11:11:25 Writing 4096 bytes at c000...
11:11:27 Writing 4096 bytes at d000...
11:12:40 tilib: MSP430_Memory: Could not write device memory (error = 7)
11:12:40 titext: data error on line 523
11:12:40 MSP430_Run
11:12:41 MSP430_Close
11:12:41 SSH: EXEC: completed after 88,479 ms
11:12:41 SSH: Disconnecting configuration [test_stand_7] ...
11:12:41 ERROR: Exception when publishing, exception message [Exec exit status not zero. Status [255]]
11:12:41 Build step 'Send files or execute commands over SSH' changed build result to UNSTABLE
11:12:41 [34000-134_Soteria_VID_Slice_Load] $ /bin/sh -xe /tmp/jenkins3901141079429766413.sh
11:12:41 + rm -rf /var/lib/jenkins/workspace/34000-134_Soteria_VID_Slice_FW_CI_CD/svn_src
11:12:41 Finished: UNSTABLE

dlbeer, 2019-06-10:
Shravan Bommu <notifications@github.com> writes:
> 11:11:24 Device: MSP430F5152
> 11:11:24 Writing  128 bytes at 1800...
> 11:11:25 Writing 4096 bytes at c000...
> 11:11:27 Writing 4096 bytes at d000...
> 11:12:40 tilib: MSP430_Memory: Could not write device memory (error = 7)
> 11:12:40 titext: data error on line 523

That appears to be an in-range write for that device.

This looks like a bug in libmsp430.so/MSP430.DLL. Have you tried using
the latest version of the TI library? Failing that, you might need to
report the bug to TI.

Cheers,
Daniel

-- 
Daniel Beer <dlbeer@gmail.com> http://dlbeer.co.nz/
PGP: BA6E 0B26 1F89 246C E3F3  C910 1E58 C43A 160A 553B