dlbeer/mspdebug #85: No unused FET found

undyingsky, 2019-06-28:
Hi,
When i try to fix issue "tilib : libmsp430.so not found", I have issue 
`hanhpham@hanhpham-desktop:~$ mspdebug tilib
MSPDebug version 0.25 - debugging tool for MSP430 MCUs
Copyright (C) 2009-2017 Daniel Beer <dlbeer@gmail.com>
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Chip info database from MSP430.dll v3.13.0.601 Copyright (C) 2013 TI, Inc.

Using new (SLAC460L+) API
MSP430_GetNumberOfUsbIfs
No unused FET found.
`
I'm running on Ubuntu 18.04. 
Any advice for me? Thanks!

dlbeer, 2019-06-28:
undyingsky <notifications@github.com> writes:
> When i try to fix issue "tilib : libmsp430.so not found", I have issue
> `hanhpham@hanhpham-desktop:~$ mspdebug tilib
> MSPDebug version 0.25 - debugging tool for MSP430 MCUs
> Copyright (C) 2009-2017 [...]

What hardware are you using?

Have you tried "sudo mspdebug tilib"?

-- 
Daniel Beer <dlbeer@gmail.com> http://dlbeer.co.nz/
PGP: BA6E 0B26 1F89 246C E3F3  C910 1E58 C43A 160A 553B

undyingsky, 2019-06-29:
i'm using msp430g2553.
i tried "sudo mspdebug tilib" and got same issue.

dlbeer, 2019-06-29:
undyingsky <notifications@github.com> writes:
>    i'm using msp430g2553.
>    i tried "sudo mspdebug tilib" and got same issue.

Sorry, I meant what programmer hardware are you using. Could you also
show the output of:

    sudo mspdebug --usb-list

Cheers,
Daniel

-- 
Daniel Beer <dlbeer@gmail.com> http://dlbeer.co.nz/
PGP: BA6E 0B26 1F89 246C E3F3  C910 1E58 C43A 160A 553B

undyingsky, 2019-06-29:
i install contiki for msp430. and i dont know what programmer hardware am  i using
Output of : sudo mspdebug --usb-list
hanhpham@hanhpham-desktop:~$ sudo mspdebug --usb-list
Devices on bus 002:
    002:001 1d6b:0003  [serial: 0000:00:14.0]
Devices on bus 001:
    001:004 0451:f432 eZ430-RF2500 [serial: 29FF4A55BBC73012]
    001:003 0c45:7603  [serial: ?]
    001:002 093a:2510 
    001:001 1d6b:0002  [serial: 0000:00:14.0]

dlbeer, 2019-06-29:
undyingsky <notifications@github.com> writes:
> i install contiki for msp430. and i dont know what programmer hardware am  i using
> Output of : sudo mspdebug --usb-list
> hanhpham@hanhpham-desktop:~$ sudo mspdebug --usb-list
> Devices on bus 002: [...]

Ok, try this:

    sudo mspdebug rf2500

Cheers,
Daniel

-- 
Daniel Beer <dlbeer@gmail.com> http://dlbeer.co.nz/
PGP: BA6E 0B26 1F89 246C E3F3  C910 1E58 C43A 160A 553B

undyingsky, 2019-06-30:
with sudo mspdebug rf2500, it's work :
`hanhpham@hanhpham-desktop:~$ sudo mspdebug rf2500
MSPDebug version 0.25 - debugging tool for MSP430 MCUs
Copyright (C) 2009-2017 Daniel Beer <dlbeer@gmail.com>
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Chip info database from MSP430.dll v3.13.0.601 Copyright (C) 2013 TI, Inc.

Trying to open interface 1 on 004
rf2500: warning: can't detach kernel driver: No data available
Initializing FET...
FET protocol version is 30394216
Set Vcc: 3000 mV
Configured for Spy-Bi-Wire
fet: FET returned error code 4 (Could not find device or device not supported)
fet: command C_IDENT1 failed
Using Olimex identification procedure
Device ID: 0x2553
  Code start address: 0xc000
  Code size         : 16384 byte = 16 kb
  RAM  start address: 0x200
  RAM  end   address: 0x3ff
  RAM  size         : 512 byte = 0 kb
Device: MSP430G2xx3
Number of breakpoints: 2
fet: FET returned NAK
warning: device does not support power profiling
Chip ID data:
  ver_id:         5325
  ver_sub_id:     0000
  revision:       00
  fab:            60
  self:           0000
  config:         00
  fuses:          00
Device: MSP430G2xx3

Available commands:
    !               fill            power           setwatch_r      
    =               gdb             prog            setwatch_w      
    alias           help            read            simio           
    blow_jtag_fuse  hexout          regs            step            
    break           isearch         reset           sym             
    cgraph          load            run             verify          
    delbreak        load_raw        save_raw        verify_raw      
    dis             md              set             
    erase           mw              setbreak        
    exit            opt             setwatch        

Available options:
    color                       gdb_loop                    
    enable_bsl_access           gdbc_xfer_size              
    enable_fuse_blow            iradix                      
    enable_locked_flash_access  lowercase_dis               
    fet_block_size              quiet                       
    gdb_default_port            

Type "help <topic>" for more information.
Use the "opt" command ("help opt") to set options.
Press Ctrl+D to quit.

(mspdebug) 
`

but i can't upload contiki launchpad
`hanhpham@hanhpham-desktop:~/contiki-launchpad/examples/launchpad/blink$ make blink.upload
msp430-objcopy blink.launchpad -O ihex blink.hex
--------------------------
Erasing msp430
mspdebug tilib "erase"
MSPDebug version 0.25 - debugging tool for MSP430 MCUs
Copyright (C) 2009-2017 Daniel Beer <dlbeer@gmail.com>
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Chip info database from MSP430.dll v3.13.0.601 Copyright (C) 2013 TI, Inc.

Using new (SLAC460L+) API
MSP430_GetNumberOfUsbIfs
No unused FET found.
../../../platform/launchpad/Makefile.launchpad:115: recipe for target 'blink.upload' failed
make: *** [blink.upload] Error 255
`

dlbeer, 2019-06-30:
undyingsky <notifications@github.com> writes:
> but i can't upload contiki launchpad
> `hanhpham@hanhpham-desktop:~/contiki-launchpad/examples/launchpad/blink$ make blink.upload
> msp430-objcopy blink.launchpad -O ihex blink.hex
> --------------------------
> Erasing msp430
> mspdebug tilib "erase"

That's still using the "tilib" driver. It looks like you'll have to
modify the Makefile in that case to get it working using the rf2500
driver instead.

Cheers,
Daniel

-- 
Daniel Beer <dlbeer@gmail.com> http://dlbeer.co.nz/
PGP: BA6E 0B26 1F89 246C E3F3  C910 1E58 C43A 160A 553B

undyingsky, 2019-06-30:
Thank you very much for your support.
I fixed this issue by using rf2500.

closed 2019-06-30 by undyingsky