dlbeer/mspdebug #96: [OrangePI] gpio initialization failed on MSP430F1232

vinibali, 2020-06-28:
Hello there!

I tried to connect to a [BMS](https://en.wikipedia.org/wiki/Battery_management_system) which has a MSP430F1232 onboard.
I think more or less I've figured out the pinout for the pcb and mapped it to the GPIO traces on my OrangePI using the [WiringPI/WiringOP](https://github.com/zhaolei/WiringOP) implementation for that SBC.

```
~$ gpio readall
 +-----+-----+----------+------+---+-Orange Pi+---+---+------+---------+-----+--+
 | BCM | wPi |   Name   | Mode | V | Physical | V | Mode | Name     | wPi | BCM |
 +-----+-----+----------+------+---+----++----+---+------+----------+-----+-----+
 |     |     |     3.3v |      |   |  1 || 2  |   |      | 5v       |     |     |
 |  12 |   8 |    SDA.0 |  OUT | 1 |  3 || 4  |   |      | 5V       |     |     |
 |  11 |   9 |    SCL.0 |  OUT | 1 |  5 || 6  |   |      | 0v       |     |     |
 |   6 |   7 |   GPIO.7 | ALT3 | 0 |  7 || 8  | 0 | ALT3 | TxD3     | 15  | 13  |
 |     |     |       0v |      |   |  9 || 10 | 0 | ALT3 | RxD3     | 16  | 14  |
 |   1 |   0 |     RxD2 |  OUT | 1 | 11 || 12 | 0 | ALT3 | GPIO.1   | 1   | 110 |
 |   0 |   2 |     TxD2 |   IN | 0 | 13 || 14 |   |      | 0v       |     |     |
 |   3 |   3 |     CTS2 |  OUT | 1 | 15 || 16 | 0 | ALT3 | GPIO.4   | 4   | 68  |
 |     |     |     3.3v |      |   | 17 || 18 | 0 | ALT3 | GPIO.5   | 5   | 71  |
 |  64 |  12 |     MOSI | ALT3 | 0 | 19 || 20 |   |      | 0v       |     |     |
 |  65 |  13 |     MISO | ALT3 | 0 | 21 || 22 | 0 | OUT  | RTS2     | 6   | 2   |
 |  66 |  14 |     SCLK | ALT3 | 0 | 23 || 24 | 0 | ALT3 | CE0      | 10  | 67  |
 |     |     |       0v |      |   | 25 || 26 | 0 | ALT3 | GPIO.11  | 11  | 21  |
 |  19 |  30 |    SDA.1 | ALT3 | 0 | 27 || 28 | 0 | ALT3 | SCL.1    | 31  | 18  |
 |   7 |  21 |  GPIO.21 |  OUT | 1 | 29 || 30 |   |      | 0v       |     |     |
 |   8 |  22 |  GPIO.22 |   IN | 0 | 31 || 32 | 0 | ALT3 | RTS1     | 26  | 200 |
 |   9 |  23 |  GPIO.23 |  OUT | 0 | 33 || 34 |   |      | 0v       |     |     |
 |  10 |  24 |  GPIO.24 |  OUT | 1 | 35 || 36 | 0 | ALT3 | CTS1     | 27  | 201 |
 |  20 |  25 |  GPIO.25 | ALT3 | 0 | 37 || 38 | 0 | ALT3 | TxD1     | 28  | 198 |
 |     |     |       0v |      |   | 39 || 40 | 0 | ALT3 | RxD1     | 29  | 199 |
 +-----+-----+----------+------+---+----++----+---+------+----------+-----+-----+
 | BCM | wPi |   Name   | Mode | V | Physical | V | Mode | Name     | wPi | BCM |
 +-----+-----+----------+------+---+-Orange Pi+---+------+----------+-----+-----+
~ $ sudo mspdebug -j -d "tdi=0 tdo=1 tms=2 tck=3 rst=4 tst=12" gpio

MSPDebug version 0.25 - debugging tool for MSP430 MCUs
Copyright (C) 2009-2017 Daniel Beer <dlbeer@gmail.com>
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Chip info database from MSP430.dll v3.3.1.4 Copyright (C) 2013 TI, Inc.

gpio tms= 2
gpio tdi= 0
gpio tdo= 1
gpio tck= 3
gpio rst= 4
gpio tst= 12
Starting JTAG
jtag_get_device: timed out
jtag_init: invalid jtag_id: 0x00
JTAG_power on
JTAG_connct 
JTAG ID: 0x00
pif: unexpected JTAG ID: 0x00
jtag_set_instruction_fetch: failed
gpio: initialization failed
led green
```
However this was the first time I used the GPIO ports on that board, I was pretty sure that was working correctly, beause I had another try with a simple LED test.
According to the original datasheet I've used TDO for P1.6 and TDI for TP1.7 pins. (Is there a way to use TA2 for eighter TDI or TDO?).
RPI has a BCM SoC while the OPI has an Allwinner chip, but the GPIO driver is mainlined to the Linux kernel for a while.
I use a somewhat fresh Arch Linux ARM on the board.
Is the gpio driver universal enough to use it one more platforms, have you ever tried it?

Regards

dlbeer, 2020-06-29:
On Sun, Jun 28, 2020 at 10:06:15AM -0700, vinibali wrote:
> I tried to connect to a [BMS](https://en.wikipedia.org/wiki/Battery_management_system) which has a MSP430F1232 onboard.
> I think more or less I've figured out the pinout for the pcb and mapped it to [...]

The GPIO JTAG driver should work with any modern Linux kernel that
exposes GPIOs via sysfs.

I'm not sure what the two different sets of numbers in the table above
refer to, but the numbers you should be using in the argument to -d
should correspond to the numbering used for /sys/class/gpio/gpio%d
directories. I would double-check this first.

-- 
Daniel Beer <dlbeer@gmail.com> http://dlbeer.co.nz/
PGP: BA6E 0B26 1F89 246C E3F3  C910 1E58 C43A 160A 553B

vinibali, 2020-06-29:
The numbers on the first column are mapped to the chip manufacturer's(**B**road**C**o**M**) output, while the second one is the path according to the WiringPI implementation(this is what I used).

dlbeer, 2020-06-29:
On Mon, Jun 29, 2020 at 09:21:38AM -0700, vinibali wrote:
> The numbers on the first column are mapped to the chip manufacturer's(**B**road**C**o**M**) output, while the second one is the path according to the WiringPI implementation(this is what I used).

Ok, so you'll need to figure out which of those two correspond to
Linux's GPIO numbering. It might help to test manually at the shell
using the out/direction files in /sys/class/gpio%d and a meter on a pin.

-- 
Daniel Beer <dlbeer@gmail.com> http://dlbeer.co.nz/
PGP: BA6E 0B26 1F89 246C E3F3  C910 1E58 C43A 160A 553B

vinibali, 2020-06-30:
I forgot that OrangePI's GPIO is pin to pin compatible with Raspberry.
So I had a look on the parameters and the changes on WiringPI's output, I pasted in before. After that was pretty clear, because the named BCM pins were modified. WiringPi is just a wrapper (and pack of libraries) to all the different GPIO wirings.
Thanks for the hint!
[image]

closed 2020-06-30 by vinibali