dlbeer/mspdebug #97: tilib on OpenBSD

hosewiejacke, 2020-06-29:
Hello all,

I'd like to use an FRAM launchpad on OpenBSD. AFAIK `tilib` is an appropriate driver for this. However, I couldn't make it work. Does anybody know what the problem could be?

`mspdebug` is compiled from #e65bc69:

	$ mspdebug --version

    MSPDebug version 0.25 - debugging tool for MSP430 MCUs
	Copyright (C) 2009-2017 Daniel Beer <dlbeer@gmail.com>
	This is free software; see the source for copying conditions.  There is NO
	warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
	Chip info database from MSP430.dll v3.15.0.1 Copyright (C) 2013 TI, Inc.

`libmsp430.so` is compiled from Rev 3.15.1.001:

	$ ldd libmsp430.so | awk '{print $NF}'

	libmsp430.so:
	Name
	/var/src/libmsp430/libmsp430.so
	/usr/local/lib/libboost_filesystem-mt.so.10.0
	/usr/local/lib/libboost_system-mt.so.10.0
	/usr/local/lib/libboost_thread-mt.so.10.0
	/usr/local/lib/libhidapi-libusb.so.0.0
	/usr/lib/libc++.so.4.0
	/usr/lib/libc++abi.so.2.1
	/usr/lib/libpthread.so.26.1
	/usr/lib/libm.so.10.1
	/usr/local/lib/libusb-1.0.so.1.2
	/usr/local/lib/libiconv.so.7.0

The Launchpad is attached:

	$ usbdevs

	Controller /dev/usb0:
	...
	addr 05: 2047:0013 Texas Instruments, MSP Tools Driver
	...

and recognized by `mspdebug`:

	$ mspdebug --usb-list
	Devices on bus 000:
	...
	000:005 2047:0013  [serial: XXXXXXXXXXXXXXXX]
	...

However, the connection via `tilib` fails:

	$ mspdebug tilib

	...

	Using new (SLAC460L+) API
	MSP430_GetNumberOfUsbIfs
	No unused FET found.

A connection via `ezfet` fails, too:

	$ mspdebug ezfet

	...

	Reset communications...
	transport/cdc_acm.c: can't send data: Function not implemented
	hal_proto_send: type: 0x92

Is the anything else I could try?

PS: @dlbeer, thanks a lot for `mspdebug`!
(`rf2500` works great on OpenBSD with the old Launchpad).

dlbeer, 2020-06-29:
On Mon, Jun 29, 2020 at 07:58:08AM -0700, Hose wie Jacke wrote:
> However, the connection via `tilib` fails:
> 
> 	$ mspdebug tilib
> 
> 	...
> 
> 	Using new (SLAC460L+) API
> 	MSP430_GetNumberOfUsbIfs
> 	No unused FET found.

Have you tried using -d to specify the device explicitly? I can't
remember how the auto-detection in the library works, but it's possible
that it's Linux-specific.

-- 
Daniel Beer <dlbeer@gmail.com> http://dlbeer.co.nz/
PGP: BA6E 0B26 1F89 246C E3F3  C910 1E58 C43A 160A 553B

hosewiejacke, 2020-06-30:
>Have you tried using -d to specify the device explicitly? 

Yes, but I might have been specifying the wrong device.

There are no `/dev/ttyACM*` on OpenBSD, so I tried

- `/dev/ttyU?`
- `/dev/cuaU?`

The results are all the same:

	Using new (SLAC460L+) API
	MSP430_Initialize: /dev/ttyU0
	tilib: MSP430_Initialize: Could not find MSP-FET430UIF on specified COM
	port (error = 57)
	tilib: device initialization failed

If I understand the man pages correctly, OpenBSD supports ACM through
umodem/ucom which are presented as `/dev/cuaU?`:

- https://man.openbsd.org/umodem.4
- https://man.openbsd.org/ucom.4

However, the "has no CM over data" claim in `dmesg` makes me suspicious:

	umodem0 at uhub1 port 2 configuration 1 interface 0
		"Texas Instruments MSP Tools Driver" rev 2.00/2.00 addr 8
	umodem0: data interface 1, has no CM over data, has no break
	umodem0: status change notification available
	ucom0 at umodem0

dlbeer, 2020-06-30:
On Tue, Jun 30, 2020 at 12:57:27AM -0700, Hose wie Jacke wrote:
> Yes, but I might have been specifying the wrong device.
>
> There are no `/dev/ttyACM*` on OpenBSD, so I tried
>
> - `/dev/ttyU?`
> - `/dev/cuaU?`
>
> The results are all the same:
>
> 	Using new (SLAC460L+) API [...]

Unfortunately the diagnostics from the library aren't too descriptive.
One thing I would try is just to run as root, in case you don't have
permission for the device. Failing that, maybe try running with strace
to see what it's actually trying to do.

-- 
Daniel Beer <dlbeer@gmail.com> http://dlbeer.co.nz/
PGP: BA6E 0B26 1F89 246C E3F3  C910 1E58 C43A 160A 553B

phillbush, 2023-12-01:
Hi.

I could get a bit further on porting `libmsp430` (aka `tilib`, aka `msp430-debug-stack`) on OpenBSD.

It also fails at the `MSP430_Initialize` step. But it goes further and do not issues it could not find a FET on the specified serial port (error 57), as happened with @hosewiejacke. With the patch I applied, libmsp430 actually finds a suitable FET and starts communicating with it.

Here's the port: https://github.com/phillbush/openbsd-ports/tree/master/devel/libmsp430
The patch to the source code (and two patches to the Makefile) can be found at the `patches/` directory.

However, libmsp430 fails at the communication itself (error 35), because of an unexpected byte read from the serial port at the beginning of the communication.

I explain the error better here: https://github.com/osresearch/MSPDebugStack/issues/1

Thank you,
Lucas de Sena.

phillbush, 2023-12-02:
I fixed the port.
tilib/libmsp430 now works fine on OpenBSD.

hosewiejacke, 2023-12-03:
This is plain awesome 🥳 
Thank you, @phillbush!