dlbeer/mspdebug #98: prog command fails to erase Info A memory

thiloschulz, 2020-07-02:
I have an ELF file with .infoA section for Info A memory. I have properly unlocked the memory using

`(mspdebug) opt enable_locked_flash_access true`

However, when using the "prog" command, the InfoA section is not erased. This may be desirable behaviour, if the ELF has no section that write to Info A. But in my case, it does, which would overwrite pre-existing 1s in flash with 0s where the new data is zero, leaving Info A in some weird, mixed state.
A temporary workaround is to erase InfoA manually.

dlbeer, 2020-07-03:
On Thu, Jul 02, 2020 at 05:57:15AM -0700, Thilo Schulz wrote:
> I have an ELF file with .infoA section for Info A memory. I have properly unlocked the memory using
>
> `(mspdebug) opt enable_locked_flash_access true`
>
> However, when using the "prog" command, the [...]

Does anyone recall whether Info A is guaranteed to be occupy the same
address range on all devices? If it did, then it would be fairly easy to
work around this.

-- 
Daniel Beer <dlbeer@gmail.com> http://dlbeer.co.nz/
PGP: BA6E 0B26 1F89 246C E3F3  C910 1E58 C43A 160A 553B

thiloschulz, 2020-07-03:
> Does anyone recall whether Info A is guaranteed to be occupy the same address range on all devices?

Only across one MCU family.
These are the offsets according to some of TI's linker scripts:

INFOA:
1. msp430f2*: ORIGIN = 0x10c0, LENGTH=0x0040
2. msp430fr2*: ORIGIN = 0x1980, LENGTH=0x0080
3. msp430f1*: ORIGIN = 0x1080, LENGTH=0x0080
4. msp430f4*: ORIGIN = 0x1080, LENGTH = 0x0080
5. msp430fr4*: ORIGIN = 0x1980, LENGTH=0x0080
6. msp430f5*: ORIGIN = 0x1980, LENGTH = 0x0080
7. msp430fr5*: ORIGIN = 0x1980, LENGTH=0x0080
8. msp430f6*: ORIGIN = 0x1980, LENGTH = 0x0080
9. msp430fr6*: ORIGIN = 0x1980, LENGTH=0x0080

dlbeer, 2020-07-03:
On Fri, Jul 03, 2020 at 02:19:35AM -0700, Thilo Schulz wrote:
> Only across one MCU family.
> These are the offsets according to some of TI's linker scripts:
>
> INFOA:
> 1. msp430f2*: ORIGIN = 0x10c0, LENGTH=0x0040
> 2. msp430fr2*: ORIGIN = 0x1980, LENGTH=0x0080
> 3. [...]

Hmm. In that case I think you're probably better off just creating a
short script to erase explicitly and use "load" instead of "prog". We do
have a chipinfo database in mspdebug but it's not guaranteed to be up to
date with the latest models.

-- 
Daniel Beer <dlbeer@gmail.com> http://dlbeer.co.nz/
PGP: BA6E 0B26 1F89 246C E3F3  C910 1E58 C43A 160A 553B