dlbeer/mspdebug #100: Make fails in windows, not finding winerror.h, stdint.h

mark0b, 2020-11-11:
Just getting started with mspdebug but I had to go into sockets.c and sport.c to add winerror.h and stdint.h includes respectively.

dlbeer, 2020-11-11:
On Tue, Nov 10, 2020 at 09:15:29PM -0800, mark0b wrote:
> Just getting started with mspdebug but I had to go into sockets.c and sport.c to add winerror.h and stdint.h includes respectively.

What toolchain are you building with? Is this MinGW?

-- 
Daniel Beer <dlbeer@gmail.com> http://dlbeer.co.nz/
PGP: BA6E 0B26 1F89 246C E3F3  C910 1E58 C43A 160A 553B

mark0b, 2020-11-11:
> On Tue, Nov 10, 2020 at 09:15:29PM -0800, mark0b wrote: Just getting started with mspdebug but I had to go into sockets.c and sport.c to add winerror.h and stdint.h includes respectively.
> What [...]

Yep, using MinGW per the instructions in the [FAQ](https://dlbeer.co.nz/mspdebug/faq.html#compile_windows).

I attempted to check out the windows.h file MinGW provides and sure enough it doesn't have an include statement for winerror. I attempted to add the line in that file but wasn't getting a different result.

dlbeer, 2020-11-12:
On Wed, Nov 11, 2020 at 09:16:38AM -0800, mark0b wrote:
> Yep, using MinGW per the instructions in the [FAQ](https://dlbeer.co.nz/mspdebug/faq.html#compile_windows).
>
> I attempted to check out the windows.h file MinGW provides and sure enough it doesn't [...]

Ok. Are you able to paste here the output of "git diff HEAD" on your
working build?

-- 
Daniel Beer <dlbeer@gmail.com> http://dlbeer.co.nz/
PGP: BA6E 0B26 1F89 246C E3F3  C910 1E58 C43A 160A 553B

mark0b, 2020-11-13:
It seems that I was using version 0.23 where if I build from 0.25, I don't experience the same issues. I was however experiencing an issue where make was defaulting to cc instead of gcc but that seems to be a local issue on my end. Thanks for being available to help though!