dlbeer/mspdebug #101: mspdebug with generic USB-UART bridge to program MSP420G2553

lorforlinux, 2021-01-13:
Hi, I was working to port BSLDemo2.exe to Linux but facing some problems. 

I was wondering if it's possible to use the mspdebug tool with the generic USB-UART bridges like CP2102, CH340, PL2303, and FT232R to program MSP430G2553?

dlbeer, 2021-01-13:
On Wed, Jan 13, 2021 at 07:09:36AM -0800, Deepak Khatri wrote:
> Hi, I was working to port BSLDemo2.exe to Linux but facing some problems. 
> 
> I was wondering if it's possible to use the mspdebug tool with the generic USB-UART bridges like CP2102, CH340, PL2303, and FT232R to program MSP430G2553? 

Yes, MSPDebug supports the bootstrap loader. You will need to ensure
that the two control pins used for entry are attached to DTR/RTS, and
perhaps specify the correct entry sequence on the command-line (see
--bsl-entry-sequence).

-- 
Daniel Beer <dlbeer@gmail.com> http://dlbeer.co.nz/
PGP: BA6E 0B26 1F89 246C E3F3  C910 1E58 C43A 160A 553B

lorforlinux, 2021-01-13:
Hey @dlbeer, Thank you for the info. I got access to MSP430G2553 with FT232RL using `mspdebug rom-bsl -d /dev/ttyUSB0` and I am able to program it with the prog command in the mspdebug console. Now, I want to program the chip with file name within the command itself. I am not able to figure out how to send the file in the command itself. I tried doing  `mspdebug rom-bsl -d /dev/ttyUSB0 blinky.txt` and multiple other variations but it's not working, please let me know how to achieve this functionality. 

I can not express how happy I am after seeing the mspdebug console, I worked so hard to reinvent the wheel and ignored this wonderful car you made. You and all the mspdebug contributors are awesome, thank you so much for this wonderful tool.

dlbeer, 2021-01-13:
On Wed, Jan 13, 2021 at 12:23:58PM -0800, Deepak Khatri wrote:
> Hey @dlbeer, Thank you for the info. I got access to MSP430G2553 with FT232RL using `mspdebug rom-bsl -d /dev/ttyUSB0` and I am able to program it with the prog command in the mspdebug console. Now, [...]

Glad you like it!

Try:

    mspdebug rom-bsl -d /dev/ttyUSB0 "prog blinky.txt"

Remember the quotes.

-- 
Daniel Beer <dlbeer@gmail.com> http://dlbeer.co.nz/
PGP: BA6E 0B26 1F89 246C E3F3  C910 1E58 C43A 160A 553B

lorforlinux, 2021-01-13:
Thank you so much @dlbeer, It worked!

lorforlinux, 2021-01-14:
Hi @dlbeer, do we have an executable release of the mspdebug for windows? 
I am not able to compile it, getting these errors:

```
process_begin: CreateProcess(NULL, pkg-config --cflags libusb, ...) failed.
Makefile:53: pipe: Bad file descriptor
process_begin: CreateProcess(NULL, pkg-config --libs libusb, ...) failed.
Makefile:54: pipe: Bad file descriptor
cc  -DUSE_READLINE  -O1 -Wall -Wno-char-subscripts -ggdb -I. -Isimio -Iformats -Itransport -Idrivers -Iutil -Iui -DLIB_DIR=\"/usr/local/lib/\" -D__Windows__ -DNO_SHELLCMD -o util/btree.o -c util/btree.c
process_begin: CreateProcess(NULL, cc -DUSE_READLINE -O1 -Wall -Wno-char-subscripts -ggdb -I. -Isimio -Iformats -Itransport -Idrivers -Iutil -Iui -DLIB_DIR=\"/usr/local/lib/\" -D__Windows__ -DNO_SHELLCMD -o util/btree.o -c util/btree.c, ...) failed.
make (e=2): The system cannot find the file specified.
make: *** [Makefile:241: util/btree.o] Error 2
```

I followed the steps provided [here](https://dlbeer.co.nz/mspdebug/faq.html#compile_windows).

dlbeer, 2021-01-14:
On Thu, Jan 14, 2021 at 01:21:03AM -0800, Deepak Khatri wrote:
> Hi @dlbeer, do we have an executable release of the mspdebug for windows?
> I am not able to compile it, getting these errors:
>
> ```
> process_begin: CreateProcess(NULL, pkg-config --cflags libusb, ...) [...]

The Makefile may have been broken a bit since it was last tested on
Windows. Try this:

    make CC=gcc

-- 
Daniel Beer <dlbeer@gmail.com> http://dlbeer.co.nz/
PGP: BA6E 0B26 1F89 246C E3F3  C910 1E58 C43A 160A 553B

lorforlinux, 2021-01-15:
Thanks, @dlbeer for the help 👍 
I did the following to compile it on windows,

This is what we have now,
```
CC ?= gcc
....
....
	ifneq (, $findstring(MINGW, $(UNAME_S)))
		PORTS_CFLAGS := $(shell pkg-config --cflags libusb)
		PORTS_LDFLAGS := $(shell pkg-config --libs libusb)
		RM = rm -rf
	endif
....
....
```

I replaced it with this,
```
CC = gcc # removed ?
....
....
	ifneq (, $findstring(MINGW, $(UNAME_S)))
		PORTS_CFLAGS := $(pkg-config --cflags libusb) # removed shell
		PORTS_LDFLAGS := $(pkg-config --libs libusb) # removed shell
		RM = rm -rf
	endif
....
....
```

for the execution I copied some dll file from energia's mspdebug folder and some as described [here](https://dlbeer.co.nz/mspdebug/faq.html#compile_windows) and it worked!

lorforlinux, 2021-01-15:
I have created a new flashing tool called [msptool](https://github.com/upsidedownlabs/msptool) for flashing MSP430G2xx3 using your awesome tool. 
This allows us to create an Arduino-like board based on the famous MSP430G2553 with Energia support!