dlbeer/mspdebug #102: MSP430g2553 doesn't ork

MarcoAAG, 2021-02-25:
Hi, Im triying to debug on msp430g2553
[image]
with this command
$sudo mspdebug -U 001:051 RF2500
I got this error
"MSPDebug version 0.22 - debugging tool for MSP430 MCUs
Copyright (C) 2009-2013 Daniel Beer <dlbeer@gmail.com>
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

rf2500: failed to open RF2500 device"

dlbeer, 2021-02-25:
On Thu, Feb 25, 2021 at 02:33:21PM -0800, Marco Aguilar wrote:
> Hi, Im triying to debug on msp430g2553
> [image]
> with this command
> $sudo mspdebug -U 001:051 RF2500
> I got this error
> "MSPDebug version 0.22 - debugging tool for MSP430 MCUs
> Copyright (C) 2009-2013 [...]

That's not an RF2500 device. You'll need to use the "tilib" driver (and
install libmsp430.so if you don't have it already).

-- 
Daniel Beer <dlbeer@gmail.com> http://dlbeer.co.nz/
PGP: BA6E 0B26 1F89 246C E3F3  C910 1E58 C43A 160A 553B

MarcoAAG, 2021-02-26:
Now i received an error when i try to compile the libmsp430.so
[image]

dlbeer, 2021-02-26:
On Thu, Feb 25, 2021 at 04:39:02PM -0800, Marco Aguilar wrote:
> Now i received an error when i try to compile the libmsp430.so
> [image]

Unfortunately I can't help you with that -- it's TI's library, so you'll
need to ask them for support, unless you can find a suitable precompiled
binary somewhere.

-- 
Daniel Beer <dlbeer@gmail.com> http://dlbeer.co.nz/
PGP: BA6E 0B26 1F89 246C E3F3  C910 1E58 C43A 160A 553B

MarcoAAG, 2021-02-26:
thks.. greetings

closed 2021-02-26 by MarcoAAG