dlbeer/mspdebug #110: Code is not running after programming in Ubuntu 20.04

danish-anwar-butt, 2021-09-11:
Hi,
I'm using mspdebugg to program my MSP430G2553 in Ubuntu 20.04 with FT232RL USB-UART bridge.  My connection between MSP430G2553 and FTDI bridge are given below :
FT232RL -> MSP430G2553 Pin
RXD-> 03
TXD-> 07
DTR-> 16
RTS-> 17
3.3V-> 01
GND -> 20.
The program was uploaded  (as shown in the screenshot) but not running even after resetting the device.
[image]

Regards,
Danish

dlbeer, 2021-09-12:
On Sat, Sep 11, 2021 at 05:00:15AM -0700, danish-anwar-butt wrote:
> I'm using mspdebugg to program my MSP430G2553 in Ubuntu 20.04 with FT232RL USB-UART bridge.  My connection between MSP430G2553 and FTDI bridge are given below :
> FT232RL -> MSP430G2553 Pin
> RXD-> 03 [...]

Hi Danish,

I would check that:

  * The program has been written to flash correctly (try the "verify"
    command).

  * The device is powered and reset correctly.

  * There isn't a bug in your program. Have you had it working before?

Cheers,
Daniel

-- 
Daniel Beer ***@***.***> http://dlbeer.co.nz/
PGP: BA6E 0B26 1F89 246C E3F3  C910 1E58 C43A 160A 553B

danish-anwar-butt, 2021-09-12:
Hi Daniel,
Thanks for your response.
* Please would you guide me on how can I use the "Verify" command?
* The device is powered and reset correctly because the same procedure is working on windows 7 OS.
Regards,
Danish

dlbeer, 2021-09-12:
On Sat, Sep 11, 2021 at 10:57:58PM -0700, danish-anwar-butt wrote:
> * Please would you guide me on how can I use the "Verify" command?

Use "verify" followed by the name of your file to read back and check
the regions that would have been written.

> * The device is powered and reset correctly because the same procedure is working on windows 7 OS.

If it's the same program and it works when programmed on Windows 7, then
that suggests the programming procedure being used here isn't working.
See what happens when you try to verify it.

-- 
Daniel Beer ***@***.***> http://dlbeer.co.nz/
PGP: BA6E 0B26 1F89 246C E3F3  C910 1E58 C43A 160A 553B

danish-anwar-butt, 2021-09-12:
Hi Daniel,
I ran the verify command but it is failed to verify with following mismatch error.

> sudo mspdebug rom-bsl -d /dev/ttyUSB0 "verify blinky.txt"
MSPDebug version 0.25 - debugging tool for MSP430 MCUs
Copyright (C) 2009-2017 Daniel Beer <dlbeer@gmail.com>
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Chip info database from MSP430.dll v3.15.0.1 Copyright (C) 2013 TI, Inc.

rom_bsl: received NAK
rom_bsl: failed on command 0x1e (addr = 0x0000, len = 0x0000)
warning: rom_bsl: failed to read version
Performing mass erase...
Sending password...
Chip ID data:
  ver_id:         5325
  ver_sub_id:     0000
  revision:       00
  fab:            60
  self:           0000
  config:         00
  fuses:          00
Device: MSP430G2xx3
Verifying   78 bytes at c000...
ERROR: mismatch at c000 (read ff, expected 21)
titext: data error on line 8

dlbeer, 2021-09-13:
On Sun, Sep 12, 2021 at 03:31:09AM -0700, danish-anwar-butt wrote:
> I ran the verify command but it is failed to verify with following mismatch error.

Hi Danish,

You will need to verify immediately after programming. The rom-bsl
driver is using a mass erase to gain access, so the verify in this
example is running on a blank chip.

Cheers,
Daniel

-- 
Daniel Beer ***@***.***> http://dlbeer.co.nz/
PGP: BA6E 0B26 1F89 246C E3F3  C910 1E58 C43A 160A 553B

danish-anwar-butt, 2021-09-14:
Hi Daniel,
Sorry for being late, How can I run verify immediately after the program?
Thanks

dlbeer, 2021-09-14:
On Tue, Sep 14, 2021 at 08:29:44AM -0700, danish-anwar-butt wrote:
> Sorry for being late, How can I run verify immediately after the program?

Just pass both arguments:

    mspdebug ... "prog ..." "verify ..."

Alternatively, you can start mspdebug in interactive mode (without
specifying the prog command on the command-line) and then enter the
commands as the mspdebug prompt.

-- 
Daniel Beer ***@***.***> http://dlbeer.co.nz/
PGP: BA6E 0B26 1F89 246C E3F3  C910 1E58 C43A 160A 553B