dlbeer/mspdebug #117: Trouble flashing binary using `prog` command

fabianpedd, 2022-03-22:
I am trying to flash a binary using `mspdebug tilib prog <filename>`. However, I am getting `prog: you need to specify a filename`. How do I specify the filename when providing the `prog` command to mspdebug?

Running `mspdebug tilib` and then using the interactive command reader with `prog <filename>` works just fine.

Thanks

dlbeer, 2022-03-23:
On Tue, Mar 22, 2022 at 04:10:15PM -0700, Fabian Peddinghaus wrote:
> I am trying to flash a binary using `mspdebug tilib prog <filename>`. However, I am getting `prog: you need to specify a filename`. How do I specify the filename when providing the `prog` command to [...]

Hi Fabian,

If you're running from the shell, then remember to put quotes around
single commands with spaces in them. For example:

    mspdebug tilib "prog file.hex"

Cheers,
Daniel

-- 
Daniel Beer ***@***.***> http://dlbeer.co.nz/
PGP: BA6E 0B26 1F89 246C E3F3  C910 1E58 C43A 160A 553B

fabianpedd, 2022-03-23:
Exactly what I was looking for. Had trouble finding the correct way to specify that. Thank you

closed 2022-03-23 by fabianpedd